stereolabs/zed-ros-wrapper #793: [ERROR] [1637227837.873518845]: Object detection error: MOTION SENSORS REQUIRED

dineshrboson, 2021-11-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Installed sdk succesfully but not able to run zed object detection model with zed2 and zed2i cameras 
device nvidia agx
cuda 10.2

### Steps to Reproduce

1.nvidia agx
2.cuda 10.2
3.jetpack 4.4
...

### Expected Result

zed object detection module is not working

### Actual Result

error
[image]
[image]

### ZED Camera model

ZED2

### Environment

```shell
ros melodic 
zed latest 3.6.1
jetpack 4.4
```

Myzhar, 2021-11-18:
Hi @dineshrboson 
this is not a bug, but it seems a problem with your camera.
Please close this issue and open a support ticket by sending an email to support@stereolabs.com.
Please attach the report generated by the ZED_Diagnostic tool to the email.

dineshrboson, 2021-11-18:
no actually its not a problem with the camera
I can echo remaining topics and with the same camera I can run object detection model in another machines

Myzhar, 2021-11-18:
Have you disabled sensors fusion in the ZED wrapper?

dineshrboson, 2021-11-18:
No I didnot

dineshrboson, 2021-11-18:
can you tell me where it is so that I will check

Myzhar, 2021-11-19:
Please close this issue and open a support ticket by sending an email to support@stereolabs.com
I must ask you to perform a few tests that require private communication to not share personal information.

closed 2021-11-19 by dineshrboson