stereolabs/zed-ros-wrapper #794: multiple versions of cuda deteced

dineshrboson, 2021-11-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I had only one version of cuda but the zed diagnostics detecting it as two

### Steps to Reproduce

xavier agx
cuda 10.2
jetpack 4.4

### Expected Result

zed object detection module is not working

### Actual Result

refer to previous issue also

### ZED Camera model

ZED2

### Environment

```shell
xavier agx
cuda 10.2
jetpack 4.4
melodic
```

Myzhar, 2021-11-19:
Hi @dineshrboson 
that's a warning that does not cause problems. However, we are aware that's a false warning and we are working on a fix because we know that on Jetson it is not possible to have different versions of CUDA installed.

closed 2021-11-19 by Myzhar

dineshrboson, 2021-11-19:
Its not normal actually 
this is failing and this is causing to fail zed object module failed while launching zed camera