stereolabs/zed-ros-wrapper #796: caktin_make error

lichenTHU, 2021-11-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

catkin_make error

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

compile success.

### Actual Result

[ 89%] Generating C++ code from zed_interfaces/stop_object_detection.srv
[ 89%] Generating Python code from SRV zed_interfaces/start_object_detection
[ 90%] Generating Python code from SRV zed_interfaces/save_3d_map
[ 90%] Generating Python code from SRV zed_interfaces/toggle_led
[ 91%] Generating Python code from SRV zed_interfaces/save_area_memory
[ 92%] Generating Python code from SRV zed_interfaces/start_svo_recording
[ 93%] Generating Python code from SRV zed_interfaces/reset_odometry
[ 93%] Generating Python code from SRV zed_interfaces/stop_3d_mapping
[ 94%] Generating Python code from SRV zed_interfaces/stop_object_detection
[ 94%] Built target zed_interfaces_generate_messages_cpp
[ 95%] Generating Python srv __init__.py for zed_interfaces
[ 95%] Generating Python msg __init__.py for zed_interfaces
[ 95%] Built target zed_interfaces_generate_messages_py
[ 95%] Built target zed_interfaces_generate_messages_eus
Scanning dependencies of target zed_interfaces_generate_messages
Scanning dependencies of target ZEDNodelets
[ 95%] Built target zed_interfaces_generate_messages
[ 95%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/tools/src/sl_tools.cpp.o
[ 97%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_sync_nodelet/src/rgbd_sensor_sync.cpp.o
[ 97%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o
[ 97%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_demux_nodelet/src/rgbd_sensor_demux.cpp.o
**/home/robot/lc_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:24:83: fatal error: experimental/filesystem: No such file or directory
 #include <experimental/filesystem>  // for std::experimental::filesystem::absolute
                                                                                   ^
compilation terminated.**
zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/build.make:62: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/tools/src/sl_tools.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/tools/src/sl_tools.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 98%] Linking CXX executable /home/robot/lc_ws/devel/lib/zed_wrapper/zed_wrapper_node
CMakeFiles/zed_wrapper_node.dir/src/zed_wrapper_node.cpp.o: In function `main':
zed_wrapper_node.cpp:(.text.startup+0x46): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
zed_wrapper_node.cpp:(.text.startup+0x70): undefined reference to `ros::names::getRemappings()'
zed_wrapper_node.cpp:(.text.startup+0x122): undefined reference to `ros::this_node::getName()'
zed_wrapper_node.cpp:(.text.startup+0x141): undefined reference to `nodelet::Loader::load(std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::vector<std::string, std::allocator<std::string> > const&)'
collect2: error: ld returned 1 exit status
zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/build.make:123: recipe for target '/home/robot/lc_ws/devel/lib/zed_wrapper/zed_wrapper_node' failed
make[2]: *** [/home/robot/lc_ws/devel/lib/zed_wrapper/zed_wrapper_node] Error 1
CMakeFiles/Makefile2:4334: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/robot/lc_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:36:0,
                 from /home/robot/lc_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:25:
/usr/local/zed/include/sl/Camera.hpp:728:141: warning: ‘deprecated’ attribute directive ignored [-Wattributes]
     /*@cond SHOWHIDDEN*/SL_DEPRECATED("Use toString")/*@endcond*/String /*@cond SHOWHIDDEN*/SL_CORE_EXPORT/*@endcond*/ model2str(MODEL model)/*@cond SHOWHIDDEN*/ /*@endcond*/;
                                                                                                                                             ^
/usr/local/zed/include/sl/Camera.hpp:729:154: warning: ‘deprecated’ attribute directive ignored [-Wattributes]
     /*@cond SHOWHIDDEN*/SL_DEPRECATED("Use toString")/*@endcond*/String /*@cond SHOWHIDDEN*/SL_CORE_EXPORT/*@endcond*/ errorCode2str(ERROR_CODE errorCode)/*@cond SHOWHIDDEN*/ /*@endcond*/;
                                                                                                                                                          ^
[... 42 lines cut ...]
         Resolution camera_resolution; /**< \deprecated see CameraConfiguration::resolution, camera resolution. */
                    ^
/usr/local/zed/include/sl/Camera.hpp:3289:30: warning: ‘deprecated’ attribute directive ignored [-Wattributes]
         Rotation getRotation();
                              ^
/usr/local/zed/include/sl/Camera.hpp:6195:48: warning: ‘deprecated’ attribute directive ignored [-Wattributes]
         int textureness_confidence_threshold = 100;
                                                ^
CMakeFiles/Makefile2:2471: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

### ZED Camera model

ZED2

### Environment

```shell
GPU NVIDA 2080
CUDA: 10.2
SDK: 3.6.1
```

Myzhar, 2021-11-25:
Hi @lichenTHU 
please provide more information about your operating system and the ROS version that you are using

lichenTHU, 2021-11-25:
> Hi @lichenTHU please provide more information about your operating system and the ROS version that you are using

Thank you. Ubuntu18 + ROS Melodic

Myzhar, 2021-11-25:
```/home/robot/lc_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:24:83: fatal error: experimental/filesystem: No such file or directory```

This seems a problem with your compiler.
Can you copy and paste the log of the command `$ gcc -v`?

lichenTHU, 2021-11-25:
> `/home/robot/lc_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/src/sl_tools.cpp:24:83: fatal error: experimental/filesystem: No such file or directory`
> 
> This seems a problem with your compiler. Can you copy and paste the log of the command `$ gcc -v`?

Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.5-4ubuntu8' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.8.5 (Ubuntu 4.8.5-4ubuntu8)

Myzhar, 2021-11-25:
You are using a very old version of gcc.
Try to update it:
https://linuxize.com/post/how-to-install-gcc-compiler-on-ubuntu-18-04/

lichenTHU, 2021-11-26:
> You are using a very old version of gcc. Try to update it: https://linuxize.com/post/how-to-install-gcc-compiler-on-ubuntu-18-04/

Thang you very much! 
using the command:
sudo update-alternatives --config gcc
sudo update-alternatives --config g++
I changed the version of gcc g++ and I've solved this problem!

closed 2022-04-19 by github-actions[bot]