stereolabs/zed-ros-wrapper #797: Possible mismatch between the zed macro file and the actual dimensions

zulfiz, 2021-12-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

In the zed macro file, the height of the zed2 camera is stated as 0.033 m. Moreover, the screw offset is states 0.0. In the zed2 datasheet, the height is 0.030m, and there is some screw offset as well. I think the height parameters are replaced by the camera thickness parameter, and the screw offset is ignored.

[image]

### Steps to Reproduce

1. Measure the dimensions of the camera
2. See the values in the xacro file
3. See the discrepancy

### Expected Result

Expected to match real values

### Actual Result

I think the values are off from the real values

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 18
ROS: Melodic
```

Myzhar, 2021-12-03:
Hi @zulfiz 
the `screw_offset_x` parameter in the xacro macro file is not a vertical offset, but a horizontal offset added because the ZED2i fixing hole is not centered in the bottom of the case as it is for all the other camera models.
Regarding the vertical dimensions, I'm going to verify them.

Myzhar, 2021-12-03:
Fixed with https://github.com/stereolabs/zed-ros-wrapper/commit/ab88ea16e29d135971aea4c90a600fbce5456691

closed 2021-12-03 by Myzhar