stereolabs/zed-ros-wrapper #798: If there any way to obtain the real time-stamp when the image was shot

JunbaoWan, 2021-12-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Device ZED2
SDK version:3.5
Platform:nvidia jetson+ros melodic
What'the issue: We want to obtain the real time-stamp when the image was shot
   

### Steps to Reproduce

The same issue as :+1: https://github.com/stereolabs/zed-ros-wrapper/issues/369
any update?
We use the follow code to get timestamp,as you know, it is grab time.
Is there any way to get the real time-stamp when the image was shot?
Code:
    mZed.retrieveImage(mat_left, sl::VIEW::LEFT, sl::MEM::CPU, mMatResolVideo);
    grab_ts = mat_left.timestamp;
    ros::Time stamp_left_raw = sl_tools::slTime2Ros(grab_ts);

### Expected Result

Get the real time-stamp when the image was shot
It is urgent to us,wish your reply,thanks

### Actual Result

Can only get grab time

### ZED Camera model

ZED2

### Environment

```shell
OS:ubuntu 18.04
Arch:ARM
GPU:Nvidia jetson xavier AGX
Other info:ROS melodic
```

Myzhar, 2021-12-06:
Hi @JunbaoWan 
that's not the GRAB time, but the time of arrival of the frame in the USB3 buffer, hence that's the nearest time to the real acquisition.
The latency is about 2-3 frames because of the UVC driver buffering behavior that is not controllable.

hriihimaki, 2022-03-28:
Is there any update or fix (expected) to this issue?

Myzhar, 2022-03-28:
There will not be a fix. This is the expected behavior of the ZED SDK.
It's not possible to know the exact moment when the frame has been grabbed, but only the time at which the frame has been received by the USB controller.
You can create a statistic of the latency of the camera by grabbing images of the screen that displays a running timer.

closed 2022-03-28 by Myzhar