stereolabs/zed-ros-wrapper #799: Zed Mini IMU data is not published

mithundiddi, 2021-12-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Upgraded to latest zed sdk 3.6.1, imu data and imu data_raw are not getting published. While Pose, odom, image topics seems to be fine.

Tested both on (1) x86_64 / ubuntu 18.04 / cuda 10.2 / zed sdk 3.6.1 / master branch of this repo, with zed mini and (2) Nividia Jetson tx2 (aarch64/amd64 / ubuntu 18.04 / cuda 10.2 / jetpack 4.6 rev3 / zed sdk 3.6.1 / master branch of this repo, with zed mini.

IMU topic was working with same zed mini in older zed sdk and wrapper code. Don't remember the sdk version and wrapper tag of the working version. (The working code I had jetson TX2 was from 2019 with jetpack 3.3)

### Steps to Reproduce

1. ubuntu 18.04 with cuda 10.2
2. zed sdk 3.6.1 with cuda 10.2
3. HW zed mini 
Sry about sub-optimal description, Let me know if you need any more information.
...

### Expected Result

IMU data

### Actual Result

imu topic is not publishing data

### ZED Camera model

ZED Mini

### Environment

```shell
*pasting same from above 
x86_64 / ubuntu 18.04 / cuda 10.2 / zed sdk 3.6.1 / master branch of this repo, with zed mini
```

mithundiddi, 2021-12-08:
The issue seems to be when sensors/sync_time is set to true 

Imu data and tf (map to Odom to base_link) gets published iff one of image topics have active subscribers. 

Subscribe to imu topics alone doesn't suffice. If sync_time is set to false, they are called with different timers, so it works

Myzhar, 2021-12-08:
That's an expected behavior. If you enable frame/sensors synchronization then both topics must be subscribed

closed 2022-03-30 by Myzhar