stereolabs/zed-ros-wrapper #800: ZED connection -> CAMERA FAILED TO SETUP on cuda 11.5 ,ubuntu 20.04

RamanaBotta, 2021-12-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

after installing cuda 11.5 zed_wrapper stopped working. previously it used to work fine. we are testing on two machines but results are the same . when i run ZED_Diagnose -> results are "make sure the camera pulgged in or try another usb port".
i don't understand is this a problem of cuda or zed sdk. 
please help
Thank you
Raman

### Steps to Reproduce

roslaunch zed_wrapper zed2.launch

### Expected Result

Success 

### Actual Result

[ INFO] [1639042935.835651212]: ZED connection -> CAMERA FAILED TO SETUP

### ZED Camera model

ZED2

### Environment

```shell
OS: POP OS
ZED_SKD: 3.6
ROS: Neotic
GPU:Nvidia
```

### Anything else?

 roslaunch zed_wrapper zed2.launch 
... logging to /home/boson/.ros/log/394a9ac6-58d4-11ec-87af-bf3592a9fe5b/roslaunch-pop-os-15365.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/boson/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://pop-os:40473/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.13
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.3
 * /zed2/zed_node/depth/openni_depth_mode: False
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
[... 145 lines cut ...]
[ INFO] [1639042906.778781461]:  * [DYN] sharpness		-> 4
[ INFO] [1639042906.778881451]:  * [DYN] gamma			-> 8
[ INFO] [1639042906.778981557]:  * [DYN] auto_exposure_gain	-> ENABLED
[ INFO] [1639042906.779265687]:  * [DYN] auto_whitebalance	-> ENABLED
[ INFO] [1639042906.780945915]:  * Camera coordinate system	-> Right HANDED Z UP and X FORWARD
[ INFO] [1639042906.780974003]:  *** Opening ZED 2...
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1639042912.849360769]: ZED connection -> CAMERA FAILED TO SETUP
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1639042920.955599181]: ZED connection -> CAMERA FAILED TO SETUP
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1639042928.295849919]: ZED connection -> CAMERA FAILED TO SETUP
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1639042935.835651212]: ZED connection -> CAMERA FAILED TO SETUP
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1639042945.567493258]: ZED connection -> CAMERA FAILED TO SETUP
^[[1;2A^[[1;2A^[[1;2A^C[zed2/zed_node-3] killing on exit
[zed2/zed2_state_publisher-2] killing on exit
^C[ZED][Init][Info] Camera opening timeout reached
ZED Nodelet destroyed
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Myzhar, 2021-12-09:
Hi @RamanaBotta,
did you reinstall the ZED SDK after upgrading to CUDA 11.5?
You must use the ZED SDK v3.6.2 that has been released recently with CUDA 11.5.
Regarding the ZED ROS Wrapper, after updating the ZED SDK, you must recompile it, but first you must clear the CMake cache by deleting the `build` and `devel` folders in the root of your catkin workspace

RamanaBotta, 2021-12-09:
yes i did these steps, 
i installed cuda 11.5 then  installed ZED SDK 3.6.2 , then as well when i ran ZED_Diagnose tool
following is the result
{
    "Camera Test": {
        "ZEDCount": 0,
        "error": [
            "<b>Camera not detected</b> <br/> Make sure the camera is plugged in or try another USB 3.0 port."
        ]
    },
    "Graphics Card": {
        "deviceCount": 1,
        "deviceDriverVersion": 11050,
        "devices": [
            {
                "arch": "Ampere",
                "computeCapability": "8.6",
                "cores": 3840,
                "name": "NVIDIA GeForce RTX 3060 Laptop GPU",
                "totalMemoryBytes": "6235422720",
                "totalMemoryMB": 5946.5625
            }
        ],
        "initResult": 0,
        "valid": [
            "<b>Graphics card: </b> NVIDIA GeForce RTX 3060 Laptop GPU"
        ]
    },
    "Processor": {
        "OS": "Pop!_OS 20.04 LTS ",
        "avxSupported": true,
        "coreCount": "16",
        "cpu": " 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz",
        "datetime": "2021-12-09 03:26:11 PM",
        "hyperThreading": "1",
        "motherboard": "QX60_TLS, TGL",
        "ramCapacity": "16.545808",
        "ramUsage": "",
        "ramUsed": "5.210305",
        "threadCount": "32",
        "valid": [
            "<b>Processor: </b>  11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz",
            "<b>Motherboard: </b> QX60_TLS, TGL"
        ]
    },
    "USB": {
        "CorruptedFirmware": false,
        "MCUDetected": true,
        "USBList": [
            {
                "USBMode": 3,
                "USB_path": "/2",
                "bDescriptorType": 1,
                "bDeviceProtocol": 1,
                "bLength": 18,
                "bMaxPacketSize0": 9,
                "bNumConfigurations": 1,
                "bcdDevice": "1.0",
                "bcdUSB": "3.0",
                "bcdUSBClass": 239,
                "bcdUSBSubClass": 2,
                "busNumber": 4,
                "device": "ZED2",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 0,
                "idProduct": "0xf780",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/9",
                "idProduct": "0xa061",
                "idVendor": "0x0408"
            },
            {
                "USBMode": 2,
                "USB_path": "/4/2",
                "bDescriptorType": 1,
                "bDeviceProtocol": 0,
                "bLength": 18,
                "bMaxPacketSize0": 64,
                "bNumConfigurations": 1,
                "bcdDevice": "3.8",
                "bcdUSB": "2.0",
                "bcdUSBClass": 0,
                "bcdUSBSubClass": 0,
                "busNumber": 3,
                "device": "ZED2 MCU",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 3,
                "idProduct": "0xf781",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/4",
                "idProduct": "0x2512",
                "idVendor": "0x0424"
            },
            {
                "USB_path": "/14",
                "idProduct": "0x0026",
                "idVendor": "0x8087"
            }
        ],
        "USBMode": 3,
        "ZED Camera Module Detected": 63360,
        "ZED MCU Module Detected": 63361,
        "ZEDDetected": true,
        "notTested": [
            "Please connect a camera to test the USB connection."
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.5.119",
        "DiagnosticZEDSDKVersion": "3.6.2",
        "InstalledZEDSDKVersion": "Unknown",
        "SMI": "(SMI) :     Performance State                     : P8",
        "ai": {
            "checkAI": true,
            "cudaVersion": 11050,
            "cudnnCudartVersion": 11030,
            "cudnnVersion": 8201,
            "cudnnVersionExpected": 8201,
            "tensorVersion": 8201
        },
        "binFiles": [
            "libsl_ai.so",
            "libsl_zed.so",
            "libsl_zed_static.a"
        ],
        "cudaDeviceCount": 0,
        "cudaLoaded": true,
        "initResult": 1,
        "loaded": true,
        "resourcesFiles": [
        ],
        "resourcesFilesExpected": [
            "objects_performance_2.1"
        ],
        "valid": [
            "<b>CUDA version:</b> 11.5"
        ],
        "warning": [
            "Multiple version of CUDA has been detected. It is recommended to uninstall other versions than 11.5."
        ]
    }
}

Myzhar, 2021-12-09:
The camera is correctly enumerated by the operating system and it is available in the USB resources, so I suppose that there is something odd with your GPU driver or CUDA installation.

Can you check in the `/usr/local/` folder if there is more than a version of CUDA available?
Please also copy and paste the outputs of the commands `nvcc --version` and `nvidia-smi`.

RamanaBotta, 2021-12-09:
`$`nvcc --version``

```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0

```

``$nvidia-smi``
```
Thu Dec  9 19:58:42 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 495.29.05    Driver Version: 495.29.05    CUDA Version: 11.5     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   36C    P5    N/A /  N/A |      0MiB /  5946MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+

```

``$ls /usr/local/``
``
bin  cuda  cuda-11  cuda-11.5  etc  games  include  lib  man  sbin  share  src  zed
``

Myzhar, 2021-12-09:
Nothing odd there.
 Can you try to open the camera with a webcam application (e.g `cheese` or `guvcview`) to verify that the UVC module is correctly working?

RamanaBotta, 2021-12-10:
No I can not open zed camera  with webcam applications (e.g cheese or guvcview). what should i do now ?

Myzhar, 2021-12-10:
@RamanaBotta your problem is probably not related with the CUDA upgrade, it's almost surely an HW problem.
Discussing here is off-topic. I close the issue, please write an email to support@stereolabs.com to go further to find a solution.

closed 2021-12-10 by Myzhar