stereolabs/zed-ros-wrapper #803: Error when trying to run colcon build on docker environment

electop, 2021-12-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm trying to build a package according to the tutorial. 
[Getting Started with ROS 2 and ZED](https://www.stereolabs.com/docs/ros2/#installation) > BUILD THE PACKAGE

### Steps to Reproduce

## Procedures
1. docker pull stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04
2. xhost +si:localuser:root
3. docker run --gpus all -it --privileged -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04
4. mkdir ~/Downloads && cd ~/Downloads
5. wget https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros2_foxy.sh
6. sudo chmod 755 ./install_ros2_foxy.sh
7. bash ./install_ros2_foxy.sh
8. sudo apt install ros-foxy-cartographer \
    ros-foxy-cartographer-ros \
    ros-foxy-navigation2 \
    ros-foxy-nav2-bringup \
9. source ~/.bashrc
10. sudo apt install ros-foxy-dynamixel-sdk \
      ros-foxy-turtlebot3-msgs \
      ros-foxy-turtlebot3 \
11. cd ~/colcon_ws
12. mkdir src
13. cd src
14. git clone https://github.com/stereolabs/zed-ros2-wrapper.git
15. cd ..
16. rosdep install --from-paths src --ignore-src -r -y
17. colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release

### Expected Result

No errors

### Actual Result

```
Starting >>> zed_interfaces
Starting >>> zed_tutorial_depth
Starting >>> zed_tutorial_pos_tracking
Starting >>> zed_tutorial_video
Finished <<< zed_tutorial_depth [7.53s]                                 
Finished <<< zed_tutorial_video [7.95s]                                 
Finished <<< zed_tutorial_pos_tracking [9.50s]                          
Finished <<< zed_interfaces [12.6s]                        
Starting >>> zed_components
Starting >>> rviz_plugin_zed_od
Finished <<< rviz_plugin_zed_od [10.5s]                                   
--- stderr: zed_components                                
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libusb-1.0.so', needed by 'libzed_camera_component.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< zed_components [21.4s, exited with code 2]

Summary: 5 packages finished [34.1s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  4 packages not processed
```

### ZED Camera model

ZED2i

### Environment

```shell
* OS : Ubuntu 20.04
* Docker image : stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04
* ROS : ROS2 Foxy
```

### Anything else?

* uname : Linux baa7ccbf323b 5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Myzhar, 2021-12-25:
@electop you need libusb, if it's available in your docker image, maybe that it's not located in the correct path:
`'/usr/lib/x86_64-linux-gnu/libusb-1.0.so'`
Try first by installing `libusb-dev`

electop, 2021-12-25:
> @electop you need libusb, if it's available in your docker image, maybe that it's not located in the correct path: `'/usr/lib/x86_64-linux-gnu/libusb-1.0.so'` Try first by installing `libusb-dev`

Thanks for your reply.
Please check this log after installing 'libusb-dev'.
```
Starting >>> zed_interfaces
Starting >>> zed_tutorial_depth
Starting >>> zed_tutorial_pos_tracking
Starting >>> zed_tutorial_video
Finished <<< zed_tutorial_video [7.57s]                                 
Finished <<< zed_tutorial_depth [7.91s]                                 
Finished <<< zed_tutorial_pos_tracking [9.80s]                          
Finished <<< zed_interfaces [12.8s]                        
Starting >>> zed_components
Starting >>> rviz_plugin_zed_od
Finished <<< rviz_plugin_zed_od [10.3s]                                   
--- stderr: zed_components                                
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libusb-1.0.so', needed by 'libzed_camera_component.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< zed_components [21.3s, exited with code 2]

Summary: 5 packages finished [34.2s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  4 packages not processed
```

Myzhar, 2021-12-25:
Is the lib available here
`'/usr/lib/x86_64-linux-gnu/libusb-1.0.so'`
If not you must search where it's located and create a symbolic link to that path

electop, 2021-12-25:
> Is the lib available here `'/usr/lib/x86_64-linux-gnu/libusb-1.0.so'` If not you must search where it's located and create a symbolic link to that path

Sure. I think so.
```
root@baa7ccbf323b:/usr/lib/x86_64-linux-gnu# ls libusb*
libusb-0.1.so.4      libusb-1.0.so.0      libusb.a   libusbmuxd.so.6
libusb-0.1.so.4.4.4  libusb-1.0.so.0.2.0  libusb.so  libusbmuxd.so.6.0.0
```

electop, 2021-12-25:
This issue resolved with his help.

closed 2021-12-25 by electop