stereolabs/zed-ros-wrapper #804: Zed in Ros

kankanzheli, 2022-01-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

How to use zed camera with Instance segmentation method like mmdetection?
Besides, can the zed camera run in the gazebo virtual environment now?

Myzhar, 2022-01-05:
@kankanzheli I transferred the issue to the correct repository.
The ZED ROS wrapper only supports bounding box extraction, with no segmentation.
There is no support for Gazebo yet because the ZED SDK does not support image frame injection from external sources.

kankanzheli, 2022-01-05:
So the zed ROS can't output the left image to the image processing tool like zed pytorch?

Myzhar, 2022-01-05:
The ROS wrapper publishes the left image and many other topics with all the information provided by the ZED SDK.
There is no support for the image segmentation feature provided by the object detection module.

kankanzheli, 2022-01-05:
That's not my problem. What I want to know is how zed ROS publishes image information and submits it to the image detection tool.

Myzhar, 2022-01-05:
The ROS wrapper publishes images, but you must write your own ROS node that subscribes to image topics and use them with the image detection toll that you are speaking about.
You need a ROS node that executes the tasks that you need

kankanzheli, 2022-01-05:
Is there no relevant code to refer to? Which is to use the zed ROS node to subscribe to image topics and provide them to image detection tools.

Myzhar, 2022-01-05:
What knowledge do you have of ROS?
You can start from our online documentation:
https://www.stereolabs.com/docs/ros/

kankanzheli, 2022-01-05:
As you said, We need Creating a Stereo Camera in Gazebo instead of Zed Camera. I want to use zed mini camera. What camera parameters need to be specified?

kankanzheli, 2022-01-06:
？？

Myzhar, 2022-01-06:
You must retrieve the camera parameters by using the ZED SDK, for example with a Python script.
Each camera has its own parameters:
https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1CalibrationParameters.html

kankanzheli, 2022-01-06:
I'm going to make a camera similar to zed Mini in gazebo. For example, there are the same model and the same camera parameters (such as the distance between camera lenses).

Not unique to my zed mini camera.

closed 2022-01-16 by kankanzheli