stereolabs/zed-ros-wrapper #806: The zed2i magnetometer value is different from the phone's magnetometer value

arenas7307979, 2022-01-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

hii, I have already calibrated the phone and zed2i magnetometer But there is a big gap value between  ​​the two devices [Ignore the direction,  have a large unit value difference]

[image]
[image]

### Steps to Reproduce

1. calibrate phone mag
2. calibrate zed2i mag  (ref from https://www.stereolabs.com/docs/sensors/magnetometer/#magnetometer-calibration)
3. Compare values ​​at same position
...

### Expected Result

 same unit value

### Actual Result

 same unit value

### ZED Camera model

ZED2i

### Environment

```shell
ubuntu 18.04
```

Myzhar, 2022-01-17:
Hi @arenas7307979
after the calibration of the magnetometer, the relative plot displays the normalized values used to estimate the camera headings that are different from the real value of the magnetic fields measured by the sensor.
To see the values of the magnetic field (similar to what the smartphone shows) you must reset your calibration.
Please note that the camera and the smartphone will never show the same values because the sensors' measures are affected by the magnetic disturbances introduced by the components close to them (hard ironing and soft ironing effects).

arenas7307979, 2022-01-18:
focus on units (normalized values to the real value of the magnetic fields)
that means do I need another tool to calibrate mag instead of using the tools provided by ZED ?

Myzhar, 2022-01-18:
Hi @arenas7307979 
the "real" values of the magnetic field are the raw values read by the sensor.
The calibration is used to compensate for the disturbances introduced by the metallic components and the magnetic components that are placed near the sensor.
As far as I know, this calibration only generates normalized values.
Please correct me if I have not understood what you are searching for.

arenas7307979, 2022-01-19:
hii, @Myzhar 
how to get raw magnetic field values from zed2i
I try to use ROS-API  to get the magnetic_field_uncalibrated  value that seems unusually small (3 axis norm about 2.5)

[image]
[image]

Myzhar, 2022-01-19:
As I told you, you can simply open the "ZED_Sensor_Viewer" application and reset the magnetometer calibration.
In this way, the magnetic_field published by ROS will contain the RAW values.
In a future update, we can add the RAW values directly as a new topic.

arenas7307979, 2022-01-19:
Thanks A Million.

closed 2022-01-19 by arenas7307979