stereolabs/zed-ros-wrapper #807: ZED connection -> CAMERA NOT DETECTED

YouShaoze, 2022-01-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cumt-Nuvo-7000-Series:34449/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /zedm/zed_node/auto_exposure_gain: True
 * /zedm/zed_node/auto_whitebalance: True
 * /zedm/zed_node/brightness: 4
 * /zedm/zed_node/contrast: 4
 * /zedm/zed_node/depth/depth_downsample_factor: 1.0
 * /zedm/zed_node/depth/depth_stabilization: 1
 * /zedm/zed_node/depth/max_depth: 10.0
 * /zedm/zed_node/depth/min_depth: 0.35
 * /zedm/zed_node/depth/openni_depth_mode: False
 * /zedm/zed_node/depth/quality: 1
 * /zedm/zed_node/depth/sensing_mode: 0
 * /zedm/zed_node/depth_confidence: 50
 * /zedm/zed_node/depth_texture_conf: 100
 * /zedm/zed_node/exposure: 100
 * /zedm/zed_node/gain: 100
 * /zedm/zed_node/gamma: 8
 * /zedm/zed_node/general/base_frame: base_link
 * /zedm/zed_node/general/camera_flip: False
 * /zedm/zed_node/general/camera_model: zedm
 * /zedm/zed_node/general/camera_name: zedm
 * /zedm/zed_node/general/gpu_id: -1
 * /zedm/zed_node/general/grab_frame_rate: 100
 * /zedm/zed_node/general/resolution: 3
 * /zedm/zed_node/general/self_calib: True
 * /zedm/zed_node/general/serial_number: 0
 * /zedm/zed_node/general/svo_compression: 2
 * /zedm/zed_node/general/verbose: False
 * /zedm/zed_node/general/zed_id: 0
 * /zedm/zed_node/hue: 0
 * /zedm/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zedm/zed_node/mapping/mapping_enabled: False
 * /zedm/zed_node/mapping/max_mapping_range: -1
 * /zedm/zed_node/mapping/resolution: 0.05
 * /zedm/zed_node/point_cloud_freq: 100.0
 * /zedm/zed_node/pos_tracking/area_memory: True
 * /zedm/zed_node/pos_tracking/area_memory_db_path: zed_area_memory.area
 * /zedm/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zedm/zed_node/pos_tracking/floor_alignment: False
 * /zedm/zed_node/pos_tracking/imu_fusion: True
 * /zedm/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zedm/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zedm/zed_node/pos_tracking/map_frame: map
 * /zedm/zed_node/pos_tracking/odometry_frame: odom
 * /zedm/zed_node/pos_tracking/path_max_count: -1
 * /zedm/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zedm/zed_node/pos_tracking/pos_tracking_enabled: True
 * /zedm/zed_node/pos_tracking/publish_map_tf: True
 * /zedm/zed_node/pos_tracking/publish_tf: True
 * /zedm/zed_node/pos_tracking/save_area_memory_db_on_exit: False
 * /zedm/zed_node/pos_tracking/two_d_mode: False
 * /zedm/zed_node/pub_frame_rate: 100.0
[... 130 lines cut ...]
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1642921855.156415458]: ZED connection -> CAMERA NOT DETECTED
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1642921862.592708757]: ZED connection -> CAMERA NOT DETECTED
[ZED][Init][Info] Camera opening timeout reached
[ INFO] [1642921869.775637993]: ZED connection -> CAMERA NOT DETECTED

### ZED Camera model

ZED Mini

### Environment

```shell
ubuntu 18.04.6
i7-8700
GTX 1660Ti
v3.6.4
ros melodic
cuda 11.2
```

### Anything else?

no

YouShaoze, 2022-01-23:
on windows 10 I can open zedmini but need waiting a long time, the camera firmware showed 1523 but the imu firmware is -1.
I tried update imu firmware but SDK showed waiting for device to reboot and unable to detect device to update, Drivers are probably missing. Make sure your PC is connected to internet for automatic driver update.

YouShaoze, 2022-01-23:
{
    "Camera Test": {
        "ZEDCount": 1,
        "camera": {
            "Device ID": 0,
            "Firmware": 1523,
            "Init Output": "OK",
            "Model": "ZED Mini",
            "Serial Number": 10027442,
            "valid": [
                "<b>Camera: </b>ZED Mini",
                "<b>Firmware: </b>1523"
            ]
        },
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initialization": "OK",
                "status": true,
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 1080,
                "height (output)": 1080,
                "initialization": "OK",
                "status": true,
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initialization": "OK",
                "status": true,
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 720,
                "height (output)": 720,
                "initialization": "OK",
                "status": true,
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@30": {
                "fps (input)": 30,
                "fps (output)": 27,
                "height (input)": 720,
                "height (output)": 720,
                "initialization": "OK",
                "status": true,
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@60": {
                "dropCount": -1,
                "fps (input)": 60,
                "fps (output)": 58,
                "height (input)": 720,
[... 276 lines cut ...]
            "cudnn_cnn_train64_8.dll",
            "cudnn_ops_infer64_8.dll",
            "cudnn_ops_train64_8.dll",
            "nvinfer.dll",
            "nvinfer_builder_resource.dll",
            "nvinfer_plugin.dll",
            "nvonnxparser.dll",
            "nvparsers.dll",
            "sl_ai64.dll",
            "sl_zed64.dll"
        ],
        "driverVersion": "511.23",
        "drivervDllVersion": "30.0.15.1123",
        "resourcesFiles": [
        ],
        "resourcesFilesExpected": [
            "objects_performance_2.1"
        ],
        "runtimeDllVersion": 11050,
        "valid": [
            "<b>ZED SDK version:</b> 3.6.3",
            "<b>CUDA version:</b> V11.5.50"
        ]
    }
}

Myzhar, 2022-01-23:
Hi @YouShaoze 
This is not a ROS bug, but a support request. 
Please write an email to support@stereolabs.com describing the problem and attaching the report file generated by the ZED Diagnostic tool.

closed 2022-01-23 by Myzhar