stereolabs/zed-ros-wrapper #810: Image sometimes flickers

YuxuanSnow, 2022-02-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi, 
while i am testing the zedm camera, I found this issue: the images sometimes flicker.

I put the camera upwards. The camera stayed static during the recording. As you can see in 3 images, frame 240 is the same as frame 242, while frame 241 looks really storage.

[image]
[image]

[image]

I tried to record multiple sequences, and neither of them get rid of this issue.

### Steps to Reproduce

1. launch ROS driver
2. all topics about images (RGB, grayscale) have this issue.

### Expected Result

Normal images without flickering

### Actual Result

Image sequnces with flickering

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 18.04
SDK:3.6.5
```

Myzhar, 2022-02-19:
Hi @YuxuanSnow 
this can be a problem of your camera. Please send an email to support@stereolabs.com and add the report file generated by the ZED_Diagnostic tool.

closed 2022-02-19 by Myzhar

roncapat, 2023-07-05:
@YuxuanSnow how have you solved?