stereolabs/zed-ros-wrapper #812: Empty zed-ros-interfaces directory in 3.7.X release

JeremieBourque1, 2022-02-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The `zed-ros-interfaces` directory in the 3.7.X release is empty when it should contain the contents of that repo.

### Steps to Reproduce

1. Download the zip or tar.gz of the 3.7.X release
2. Check the content of the `zed-ros-interfaces` directory
...

### Expected Result

The `zed-ros-interfaces` directory should contain the contents of that repo.

### Actual Result

The directory is empty

### ZED Camera model

ZED

### Environment

```shell
Ubuntu 18.04
```

Myzhar, 2022-02-26:
The releases are automatically created by Github.
We must check why it does not populated submodules.
You can get the 'zed-ros-interfaces' repository zip file and decompress it there to make it work.

Myzhar, 2022-02-26:
I confirm that it's not yet possible to generate a release file with sub-modules source code:
https://github.community/t/create-release-that-contains-submodule/1329

closed 2022-02-28 by Myzhar