stereolabs/zed-ros-wrapper #813: Out put the data from svo file and save the pointcloud as numpy

kankanzheli, 2022-02-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Now I am try to use the zed camera with ros.
When I am not working with the ros, we can use code from python to output the data from .svo files and save those data such as left data in to rgb image. And save the pointcloud in to a numpy array with shape(720,1280,4)
By:         
def load_image_into_numpy_array(image):
    ar = image.get_data()
    ar = ar[:, :, 0:3]
    (im_height, im_width, channels) = image.get_data().shape
    return np.array(ar).reshape((im_height, im_width, 3)).astype(np.uint8)
.....
        zed.retrieve_measure(ptcloud, sl.MEASURE.XYZ, resolution=res)
        zed.retrieve_image(depth_img, sl.VIEW.DEPTH, resolution=res)
        ptcloud_np_global = np.array(ptcloud.get_data())
        image_np_global = load_image_into_numpy_array(left)

So, the first question, can we open the .svo files with ros?
The how can we save the rgb data and pointcloud data in to the numpy array which is same as the numpy array from the above code?

kankanzheli, 2022-02-28:
Forgot to say. The camera I use is zed Mini

closed 2022-02-28 by kankanzheli

Myzhar, 2022-02-28:
Hi @kankanzheli 

`So, the first question, can we open the .svo files with ros?`
You can exploit the `svo_file` parameters:
https://www.stereolabs.com/docs/ros/#launching-with-recorded-svo-video

`The how can we save the rgb data and pointcloud data in to the numpy array which is same as the numpy array from the above code?`
You must write your own Python node that subscribes to the topics that you want to export and in the callback add the conversion code.
Here's a getting-started tutorial:
http://wiki.ros.org/rospy_tutorials/Tutorials/numpy