stereolabs/zed-ros-wrapper #816: IMU data

wildpig888, 2022-03-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

IMU.NoiseGyro IMU.NoiseAcc IMU.GyroWalk IMU.AccWalk
IMU.Frequency

Can I get these data directly? Or I need to calibriate the camera.
Beside, I want to know about the Camera-IMU Transform:
0.999998 0.000336 -0.001944 -0.002000
-0.000335 1.000000 0.000816 -0.023000
0.001944 -0.000816 0.999998 -0.002000
0.000000 0.000000 0.000000 1.000000
Does it mean the transformation for IMU to camera?
Thanks for help.

Myzhar, 2022-03-03:
> Does it mean the transformation for IMU to camera?
Yes, exactly.

> IMU.NoiseGyro IMU.NoiseAcc IMU.GyroWalk IMU.AccWalk IMU.Frequency
The ROS wrapper does not provide the values, but they are available in the SDK.
You can modify the code to add this feature:
https://www.stereolabs.com/docs/api/structsl_1_1SensorParameters.html

wildpig888, 2022-03-03:
Thanks for your help.

closed 2022-03-03 by wildpig888