stereolabs/zed-ros-wrapper #818: ZED node died while operating

nickzherdev, 2022-03-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

While operating ZED node died with the following errors:

**dmesg**
> [Tue Mar  1 15:18:31 2022] usb 1-8.2: usbfs: process 20078 (nodelet) did not claim interface 0 before use
> [Tue Mar  1 15:18:31 2022] usb 1-8.2: reset full-speed USB device number 6 using xhci_hcd
> [Tue Mar  1 15:18:31 2022] hid-generic 0003:2B03:F781.0007: hiddev2,hidraw3: USB HID v1.11 Device [STEREOLABS ZED-2 HID INTERFACE] on usb-0000:00:14.0-8.2/input0

**roslog**
> [ INFO] [1646137111.764283003]: Loading nodelet /vision/front/zed of type zed_nodelets/ZEDWrapperNodelet to manager /vision/front/zed_nodelet_manager with the following remappings:
> [ INFO] [1646137507.649985107]: Bond broken, exiting
> 

### Steps to Reproduce

We were unable to reproduce this behavior

### Expected Result

-

### Actual Result

We ran ZED diagnostics right after node crash, the results are present in "Environment" field

### ZED Camera model

ZED2

### Environment

```shell
{
    "Camera Test": {
        "ZEDCount": 0,
        "error": [
            "<b>Camera not detected</b> <br/> Make sure the camera is plugged in or try another USB 3.0 port."
        ]
    },
    "Graphics Card": {
        "deviceCount": 1,
        "deviceDriverVersion": 11020,
        "devices": [
            {
                "arch": "Turing",
                "computeCapability": "7.5",
                "cores": 2816,
                "name": "GeForce GTX 1660 SUPER",
                "totalMemoryBytes": "6232014848",
                "totalMemoryMB": 5943.3125
            }
        ],
        "initResult": 0,
        "valid": [
            "<b>Graphics card: </b> GeForce GTX 1660 SUPER"
        ]
    },
    "Processor": {
        "OS": "Linux Mint 19.3 Tricia ",
        "avxSupported": true,
        "coreCount": "8",
        "cpu": " Intel(R) Core(TM) i7-9700E CPU @ 2.60GHz",
        "datetime": "2022-03-01 04:08:55 PM",
        "hyperThreading": "0",
        "motherboard": "NVS-7000, Neousys Technology Inc.",
        "ramCapacity": "16,698577",
        "ramUsage": "",
        "ramUsed": "14,603756",
        "threadCount": "8",
        "valid": [
            "<b>Processor: </b>  Intel(R) Core(TM) i7-9700E CPU @ 2.60GHz",
            "<b>Motherboard: </b> NVS-7000, Neousys Technology Inc."
        ]
    },
    "USB": {
        "CorruptedFirmware": false,
        "MCUDetected": true,
        "USBList": [
            {
                "USB_path": "/9",
                "idProduct": "0x9071",
                "idVendor": "0x1199"
            },
            {
                "USB_path": "/8/3",
                "idProduct": "0x0617",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/8/2",
                "idProduct": "0x0617",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/8/1",
                "idProduct": "0x0617",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/8",
                "idProduct": "0x0617",
                "idVendor": "0x05e3"
            },
            {
                "USBMode": 3,
                "USB_path": "/7",
                "bDescriptorType": 1,
                "bDeviceProtocol": 1,
                "bLength": 18,
                "bMaxPacketSize0": 9,
                "bNumConfigurations": 1,
                "bcdDevice": "1.0",
                "bcdUSB": "3.0",
                "bcdUSBClass": 239,
                "bcdUSBSubClass": 2,
                "busNumber": 2,
                "device": "ZED2",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 0,
                "idProduct": "0xf780",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/9/3",
                "idProduct": "0x0610",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/9/2",
                "idProduct": "0x0610",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/9/1/4",
                "idProduct": "0x0120",
                "idVendor": "0x2357"
            },
            {
                "USB_path": "/9/1",
                "idProduct": "0x0610",
                "idVendor": "0x05e3"
            },
            {
                "USB_path": "/9",
                "idProduct": "0x0610",
                "idVendor": "0x05e3"
            },
            {
                "USBMode": 2,
                "USB_path": "/8/2",
                "bDescriptorType": 1,
                "bDeviceProtocol": 0,
                "bLength": 18,
                "bMaxPacketSize0": 64,
                "bNumConfigurations": 1,
                "bcdDevice": "3.8",
                "bcdUSB": "2.0",
                "bcdUSBClass": 0,
                "bcdUSBSubClass": 0,
                "busNumber": 1,
                "device": "ZED2 MCU",
                "iManufacturer": 1,
                "iProduct": 2,
                "iSerial": 3,
                "idProduct": "0xf781",
                "idVendor": "0x2b03"
            },
            {
                "USB_path": "/8",
                "idProduct": "0x2512",
                "idVendor": "0x0424"
            },
            {
                "USB_path": "/6",
                "idProduct": "0x0001",
                "idVendor": "0x0eef"
            },
            {
                "USB_path": "/3",
                "idProduct": "0xc52b",
                "idVendor": "0x046d"
            }
        ],
        "USBMode": 3,
        "ZED Camera Module Detected": 63360,
        "ZED MCU Module Detected": 63361,
        "ZEDDetected": true,
        "notTested": [
            "Please connect a camera to test the USB connection."
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.0.221",
        "DiagnosticZEDSDKVersion": "3.5.6",
        "InstalledZEDSDKVersion": "Unknown",
        "SMI": "(SMI) :     Performance State                     : P2",
        "initResult": 1,
        "loaded": true,
        "valid": [
            "<b>CUDA version:</b> 11.0"
        ],
        "warning": [
            "Multiple version of CUDA has been detected. It is recommended to uninstall other versions than 11.0."
        ]
    }
}
```

Myzhar, 2022-03-10:
Hi @nickzherdev 
I'm sorry, but there is not enough information to understand the cause of the problem.

nickzherdev, 2022-03-10:
Hello, Walter!
Yeap, unfourtanately that are all the logs we've got.
And I forgot to mention the zed-ros-wrapper version. It's v3.5

Could you please suggest how could we get more logs when this happens again?  That was not the first time.

Myzhar, 2022-03-10:
I'm sorry, but you already used all the available and useful logs.
If the problem happened only one time maybe it's related to a very rare OS status.
Please keep me posted if it happens again and if you get more info.

closed 2022-03-28 by Myzhar

nickzherdev, 2022-03-29:
Hello, Walter!
I've got some updates with ZED 2i on another machine.
ZED was certanly connected to USB 3.0, nothing was plugged in the near USB ports. ZED_Doagnostics showed these results:
[image]

nickzherdev, 2022-03-29:
> {
    "Camera Test": {
        "ZEDCount": 1,
        "deviceInitResult": 6,
        "error": [
            "<b>Specific resolutions are not available</b> <br/> Replug your device or connect the camera to another <b>USB 3.0</b> port."
        ],
        "resolutions": {
            "HD1080@15": {
                "initCode": false,
                "initialization": "Camera already used"
            },
            "HD1080@30": {
                "initCode": false,
                "initialization": "Camera already used"
            },
            "HD2K@15": {
                "initCode": false,
                "initialization": "Camera already used"
            },
            "HD720@15": {
                "initCode": false,
                "initialization": "Camera already used"
            },
            "HD720@30": {
                "initCode": false,
                "initialization": "Camera already used"
            },
            "HD720@60": {
                "initCode": false,
                "initialization": "Camera already used"
            },
            "VGA@100": {
                "fps (input)": 100,
                "fps (output)": 15,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            },
            "VGA@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            },
            "VGA@30": {
                "fps (input)": 30,
                "fps (output)": 15,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
                "width (output)": 672
            },
            "VGA@60": {
                "fps (input)": 60,
                "fps (output)": 15,
                "height (input)": 376,
                "height (output)": 376,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 672,
[... 108 lines cut ...]
                "idVendor": "0x0eef"
            }
        ],
        "USBMode": 2,
        "ZED Camera Module Detected": 63616,
        "ZEDDetected": true,
        "error": [
            "<b>USB 2.0 connection detected</b><br/> Please connect your camera to an <b>USB 3.0</b> port."
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.0.221",
        "DiagnosticZEDSDKVersion": "3.5.6",
        "InstalledZEDSDKVersion": "Unknown",
        "SMI": "(SMI) :     Performance State                     : P8",
        "initResult": 1,
        "loaded": true,
        "valid": [
            "<b>CUDA version:</b> 11.0"
        ],
        "warning": [
            "Multiple version of CUDA has been detected. It is recommended to uninstall other versions than 11.0."
        ]
    }
}

nickzherdev, 2022-03-29:
Also, ZED_Depth_Viewer didn't work, but ZED_Explorer showed images from both cameras.

I tried plugging ZED in my notebook, and the Diagnostics results (USB error) were the same.

nickzherdev, 2022-03-29:
One more thing: I tried changing USB port on main machine (not my laptop), and surprisingly the "multiple CUDA versions error" didn't appear

Myzhar, 2022-03-30:
@nickzherdev please write an email to support@stereolabs.com 
We are out of topic here because this is a problem with your system configuration and not a ROS wrapper bug.

jmericks, 2024-03-11:
@nickzherdev were you able to find the root cause for this issue? We are experiencing a similar issue but have not been able to find where it is coming from. We see this "Bond broken, exiting" error every time we run the software, but the amount of time after launch seems to be random.