stereolabs/zed-ros-wrapper #819: Can not download model

lesun90, 2022-03-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I using Zed with Jetson Nano. And I am able to run ZED example code with no problem.

However, when I try using ROS with zed by `roslaunch zed_display_rviz display_zed.launch`

and I got this message **`Optimized model not found, trying to download it...`**

And it had been spinning for hours, I check the network usage and look like there is nothing being download.

Please assist. Thanks

### Steps to Reproduce

`roslaunch zed_display_rviz display_zed.launch`

### Expected Result

run ROS xviz

### Actual Result

**Optimized model not found, trying to download it...**

spinning for hours and nothing being downloaded

### ZED Camera model

ZED2

### Environment

```shell
jetson NANO
```

Myzhar, 2022-03-12:
@lesun90 what AI model are you using?
I suggest you use `ZED_diagnostic` to download/optimize the models before using them:
```
$ ZED_Diagnostic --help
Usage:
    -d | Auto start the diagnostic (only for GUI mode)
    -c | Start the application in command line (Useful when no GUI is available)
    -aid | Download all models for the object detection module
    -aio | Optimize (and download if needed) all models for the object detection module
    -aic | Clear all models for the object detection module
    -ais X | Download and Optimize the associated sl::DETECTION_MODEL (int)
    -nrlo | Download and Optimize the NEURAL depth mode
    -r | Recover an MCU module with not valid serial number
    -h | Display this help
```

Use the `-aio` flag with the model number.
If you are using the NEURAL depth mode do not use the `-nrlo` flag, but use `-ais 8` because there is a bug with `-nrlo` that will be fixed in v3.7.1

lesun90, 2022-03-12:
Thank for great and quickly support. Really impressed and appriciaited. 
Still not sure my the model not download from ros script. And that tool "ZED_diagnostic" is really helpful.

closed 2022-03-12 by lesun90

interqhq, 2022-07-14:
> @lesun90 what AI model are you using? I suggest you use `ZED_diagnostic` to download/optimize the models before using them:
>
> ```
> $ ZED_Diagnostic --help
> Usage:
>     -d | Auto start the diagnostic [...]

Thanks for your answer, which will help to solve my problem. 
However, what is the " model number" when I use -aio flag or where can I find the number? thanks!

Myzhar, 2022-07-14:
@interqhq never replies to a closed issue. It's better if you open a new issue and you cite the existing closed issues.

However, the AI_MODELS are defined here: https://www.stereolabs.com/docs/api/group__Object__group.html#gaf90fcebb970fc56cb9963d1a8b91a8d1
the first has index `0`, the last `8`