stereolabs/zed-ros-wrapper #823: IMU frequency unstable

YuxuanSnow, 2022-03-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi,

i have a zed mini camera, whose imu frequency should be 800 Hz according to the data sheet: https://www.generationrobots.com/media/zed-mini-camera-datasheet.pdf. However, when i use `rostopic hz` to monitor the frequency of the imu topic, it varies from 500 to 580, not stable. Is it normal?

Somebody asked a question regarding how to modify the updating rate, but it is still open: https://github.com/stereolabs/zed-ros-wrapper/issues/770

Thanks.

### Steps to Reproduce

1. `roslaunch zed_wrapper zedm.launch`

### Expected Result

1. imu rate to 800 Hz

### Actual Result

1. IMU rate only 560 Hz, and really unstable

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 20.04
CUDA: 11.3
GPU: 2080 Ti
```

Myzhar, 2022-03-28:
Added a parameter to set the sensors' publishing frequency. 
You can update to the latest master to test it

ROSNovice, 2022-04-16:
hi i also face the same problem. when i open the raw image in rviz, the camera like shaking

closed 2022-05-23 by github-actions[bot]