stereolabs/zed-ros-wrapper #824: CPU usage of the zed node keeps on increasing over time

honest-prateek, 2022-03-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

We are running 2 zed2i cameras and have observed that sometimes the when the system is idle, the CPU usage of the zed node kept on increasing gradually over time from 15% to more than 50%. This was when the system was idle.

### Steps to Reproduce

1. Run 2 zed2i cameras with the following params.

```
<launch>
    <!-- System name -->
    <arg name="system_name"  default="zed_multi" />

    <!-- Name of the Nodelet Manager -->
    <arg name="nodelet_manager_name"    default="$(arg system_name)_nodelet_manager" />

    <!-- Start Rviz? -->
    <arg name="start_rviz"              default="false" />

    <!-- LEFT CAMERA -->
    <arg name="camera_name_3"           default="zed2_left" /> <!-- The name you want -->
    <arg name="camera_model_3"          default="zed2i" /> <!-- 'zed' or 'zedm' or 'zed2' -->    
    <arg name="zed_nodelet_name_3"      default="zed_nodelet_left" />

    <!-- Base frame -->
    <arg name="base_frame_3"            default="base_link_robot" />

    <!-- Publish ZED urdf -->
    <arg name="publish_urdf_3"          default="true" />

    <!-- Publish odom TF -->
    <arg name="publish_tf_3"            default="false" /> <!-- Only one node must pushish TF, unless you use a Kalman filter to fuse odometries -->

    <arg name="camera_sn_3"             default="33032397" /> <!-- Very important to be able to identify the camera uniquely-->  
    <arg name="gpu_id_3"                default="-1" />

    <!-- Position respect to base frame (i.e. "base_link) -->
    <arg name="cam_pos_x_3"             default="0.019" />
    <arg name="cam_pos_y_3"             default="0.3035" />
    <arg name="cam_pos_z_3"             default="0.325" />

    <!-- Orientation respect to base frame (i.e. "base_link) -->
    <arg name="cam_roll_3"              default="0.0" />
    <arg name="cam_pitch_3"             default="0.0" />
    <arg name="cam_yaw_3"               default="1.57" /> <!-- Looking left -->

    <!-- RIGHT CAMERA -->
    <arg name="camera_name_4"           default="zed2_right" /> <!-- The name you want -->
    <arg name="camera_model_4"          default="zed2i" /> <!-- 'zed' or 'zedm' or 'zed2' -->    
    <arg name="zed_nodelet_name_4"      default="zed_nodelet_right" />

    <!-- Base frame -->
    <arg name="base_frame_4"            default="base_link_robot" />

    <!-- Publish ZED urdf -->
    <arg name="publish_urdf_4"          default="true" />

    <!-- Publish odom TF -->
    <arg name="publish_tf_4"            default="false" /> <!-- Only one node must pushish TF, unless you use a Kalman filter to fuse odometries -->

    <arg name="camera_sn_4"             default="35916364" /> <!-- 36353734 Very important to be able to identify the camera uniquely-->  
    <arg name="gpu_id_4"                default="-1" />

    <!-- Position respect to base frame (i.e. "base_link) -->
    <arg name="cam_pos_x_4"             default="0.019" />
    <arg name="cam_pos_y_4"             default="-0.3035" />
    <arg name="cam_pos_z_4"             default="0.325" />

    <!-- Orientation respect to base frame (i.e. "base_link) -->
    <arg name="cam_roll_4"              default="0.0" />
    <arg name="cam_pitch_4"             default="0.0" />
    <arg name="cam_yaw_4"               default="4.71" /> <!-- Looking backward -->

    <!-- Start Rviz-->
    <group if="$(eval arg('start_rviz')==true)">
        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find zed_multicamera_example)/rviz/multi_cam_single_nodelet.rviz" output="screen" />
    </group>

    <!-- Start nodelet manager and two nodelets in the "system_name" namespace-->
    <group ns="$(arg system_name)">
        <!-- Common nodelet Manager -->
        <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)"  args="manager" output="screen" />
        
        <group ns="$(arg camera_name_3)">
            <!-- Load ZED wrapper nodelet for camera 3-->
            <include file="$(find zed_multicamera_example)/launch/include/zed_camera_nodelet_mod.launch">
                <arg name="nodelet_manager_name"            value="/$(arg system_name)/$(arg nodelet_manager_name)" />
                <arg name="camera_name"                     value="$(arg camera_name_3)" />
                <arg name="node_name"                       value="$(arg zed_nodelet_name_3)" />
                <arg name="camera_model"                    value="$(arg camera_model_3)" />
                <arg name="base_frame"                      value="$(arg base_frame_3)" />
                <arg name="publish_urdf"                    value="$(arg publish_urdf_3)" />
                <arg name="publish_tf"                      value="$(arg publish_tf_3)" />
                <arg name="camera_sn"                      value="$(arg camera_sn_3)" />
                <arg name="gpu_id"                          value="$(arg gpu_id_3)" />
                <arg name="cam_pos_x"                       value="$(arg cam_pos_x_3)" />
                <arg name="cam_pos_y"                       value="$(arg cam_pos_y_3)" />
                <arg name="cam_pos_z"                       value="$(arg cam_pos_z_3)" />
                <arg name="cam_roll"                        value="$(arg cam_roll_3)" />
                <arg name="cam_pitch"                       value="$(arg cam_pitch_3)" />
                <arg name="cam_yaw"                         value="$(arg cam_yaw_3)" />                
            </include>
        </group>      
        
         <group ns="$(arg camera_name_4)">
            <!-- Load ZED wrapper nodelet for camera 4-->
            <include file="$(find zed_multicamera_example)/launch/include/zed_camera_nodelet_mod.launch">
                <arg name="nodelet_manager_name"            value="/$(arg system_name)/$(arg nodelet_manager_name)" />
                <arg name="camera_name"                     value="$(arg camera_name_4)" />
                <arg name="node_name"                       value="$(arg zed_nodelet_name_4)" />
                <arg name="camera_model"                    value="$(arg camera_model_4)" />
                <arg name="base_frame"                      value="$(arg base_frame_4)" />
                <arg name="publish_urdf"                    value="$(arg publish_urdf_4)" />
                <arg name="publish_tf"                      value="$(arg publish_tf_4)" />
                <arg name="camera_sn"                      value="$(arg camera_sn_4)" />
                <arg name="gpu_id"                          value="$(arg gpu_id_4)" />
                <arg name="cam_pos_x"                       value="$(arg cam_pos_x_4)" />
                <arg name="cam_pos_y"                       value="$(arg cam_pos_y_4)" />
                <arg name="cam_pos_z"                       value="$(arg cam_pos_z_4)" />
                <arg name="cam_roll"                        value="$(arg cam_roll_4)" />
                <arg name="cam_pitch"                       value="$(arg cam_pitch_4)" />
                <arg name="cam_yaw"                         value="$(arg cam_yaw_4)" />                
            </include>
        </group> 
    </group>
</launch>
```

### Expected Result

CPU usage should not increase in idle state.

### Actual Result

CPU usage increases gradually in idle state.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 18.04
CPU: Intel i7
GPU: Nvidia RTX 2060
ZED SDK v3.6.5
```

Myzhar, 2022-03-24:
Hi @honest-prateek 
how did you measure the CPU usage?
I see that you are running Rviz on the same machine, so I expect that it subscribes to many topics from both the ZED nodes.
50% of global CPU usage is expected

Myzhar, 2022-03-24:
@kushal2507 thank you for the details. I'm going to investigate the problem and provide a fix as soon as possible.

Myzhar, 2022-03-28:
I verified and this is expected because even if no topics are subscribed, the node continues to grab data and to perform the positional tracking processing to publish the TF information

kushal2507, 2022-03-28:
@Myzhar Is there a way for us to prevent the camera from publishing the topics we dont need?

Myzhar, 2022-03-28:
Usually by not subscribing to a topic stops the ZED Node from processing the relative data.
In the case of the pose and odometry, the node processes the data even if the `odom` and the `pose` topics are not subscribed in order to broadcast the `map` -> `odom` -> `base_link` TF chain.
To stop this processing you can simply set the `pos_tracking_enable` parameter to false.

closed 2022-05-04 by github-actions[bot]