stereolabs/zed-ros-wrapper #828: ZED2i work well with ZED SDK samples but not on while using ros_wrapper

kamzero, 2022-04-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

My ZED2i works well with ZED SDK, and I can run all the samples normally.
However, the camera seems not to be fetched by zed_ros_warpper.

### Steps to Reproduce

1. Install SDK according to docs, run samples normally
2. Install zed_ros_wrapper according to docs, compile successfully
3. Try to run some samples in ROS, such as `roslaunch zed_wrapper zed2i.launch`

### Expected Result

Maybe I could see the images.

### Actual Result

After `roslaunch zed_wrapper zed2i.launch`, the light of camera is on. However, there is no new output in the terminal other than: `[ INFO] [1648810295.876312577]:  *** Opening ZED 2i...`
When I kill this process, the light of camera is still on, but if I run the sdk example directly, the light will go out after the process ends. 

### ZED Camera model

ZED2i

### Environment

```shell
Ubuntu 20.04
ROS Noetic
CUDA 11.4.0
ZED SDK v3.7.1
```

Myzhar, 2022-04-01:
Hi @kamzero 
this is not a bug because I'm currently using a ZED2i with this ROS wrapper.
Please write an email to support@stereolabs.com attaching the report file generated by the ZED Diagnostic tool to receive support.

closed 2022-04-01 by Myzhar

kamzero, 2022-04-01:
Thanks for such a quick reply!