stereolabs/zed-ros-wrapper #830: custom object detection with yolov5 model

VenkataPrasadBoson, 2022-04-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

hi zed folks

i tried running my own pytorch model subscribing image  and pointcloud also processing into 3d coordinates while running i found it is running very slow then the your inbuilt detection models you have , that to in NVIDIA AGX .
so i thought of running your own custom object detector on samples which is running in real time so if possible please give any update on this or  include it on the zed wrapper so it will launch with the weights.
2) publish a document for code modification for changing these default parameters
3) try adding semantic segmentation models also 

### Use-Case

to use custom models which your inbuilt one's can't detect could be helpful for self driving car application and hope it increase the FPS output or sync with the pointcloud and bounding box data

### Anything else?

please give any update if there are any other ways i can do it

VenkataPrasadBoson, 2022-04-14:
hi @Myzhar 

sorry for the late replay 
1)ok i will try to use Nvidia TensorRT
2) about adding custom object detection including into the zed_wrapper or to do documentation for user to do customization's of complete zed wrapper if possible there may be a lot of chances to get recognition from open source community
3) Im worked with so many simulators they do publish on image msg for publishing
[image]
[image]

Myzhar, 2022-04-14:
OK, so they "simply" publish image topics with segmented objects. This can be a good feature to add.

VenkataPrasadBoson, 2022-04-19:
thanks @Myzhar i'm getting huge performance difference with it, thanks for recommendation

closed 2022-06-02 by VenkataPrasadBoson

tugbakara, 2023-05-18:
> You should do the same if you want to reach the same performance.
> 
> >

Hi @Myzhar , I want to achieve this part , I want to use ROS with ZED cam to run object detection of my custom model but where I  need to put tensorRT o/p of yolo into zed wrapper?