stereolabs/zed-ros-wrapper #836: zed_wrapper_nodelet

Amit10311, 2022-05-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

```
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:1439:55: error: ‘ANIMAL’ is not a member of ‘sl::OBJECT_CLASS’
             mObjDetFilter.push_back(sl::OBJECT_CLASS::ANIMAL);
                                                       ^~~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:1442:55: error: ‘ELECTRONICS’ is not a member of ‘sl::OBJECT_CLASS’
             mObjDetFilter.push_back(sl::OBJECT_CLASS::ELECTRONICS);
                                                       ^~~~~~~~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:1445:55: error: ‘FRUIT_VEGETABLE’ is not a member of ‘sl::OBJECT_CLASS’
             mObjDetFilter.push_back(sl::OBJECT_CLASS::FRUIT_VEGETABLE);
                                                       ^~~~~~~~~~~~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:1448:55: error: ‘SPORT’ is not a member of ‘sl::OBJECT_CLASS’
             mObjDetFilter.push_back(sl::OBJECT_CLASS::SPORT);
                                                       ^~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::publishSensData(ros::Time)’:
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:2837:79: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
     if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i) {
                                                                               ^~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:2913:79: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
     if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i) {
                                                                               ^~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::callback_updateDiagnostic(diagnostic_updater::DiagnosticStatusWrapper&)’:
/home/akumar/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:3932:79: error: ‘ZED2i’ is not a member of ‘sl::MODEL’
     if (mZedRealCamModel == sl::MODEL::ZED2 || mZedRealCamModel == sl::MODEL::ZED2i) {
                                                                               ^~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘bool zed_nodelets::ZEDWrapperNodelet::on_start_object_detection(zed_interfaces::start_object_detection::Request&, zed_interfaces::start_object_detection::Response&)’:
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:4274:106: error: ‘HUMAN_BODY_MEDIUM’ is not a member of ‘sl::DETECTION_MODEL’
     if (mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_ACCURATE || mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_MEDIUM || mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_FAST) {
                                                                                                          ^~~~~~~~~~~~~~~~~
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::processDetectedObjects(ros::Time)’:
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:4360:59: error: ‘class sl::ObjectData’ has no member named ‘sublabel’; did you mean ‘label’?
         objMsg->objects[idx].sublabel = sl::toString(data.sublabel).c_str();
                                                           ^~~~~~~~
                                                           label
/home/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:4383:110: error: ‘HUMAN_BODY_MEDIUM’ is not a member of ‘sl::DETECTION_MODEL’
         if (mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_ACCURATE || mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_MEDIUM || mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_FAST) {
                                                                                                              ^~~~~~~~~~~~~~~~~
zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:4995: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

```

### Steps to Reproduce

1. $ git clone --recursive https://github.com/stereolabs/zed-ros-wrapper.git
2. $ rosdep install --from-paths src --ignore-src -r -y
3. $ catkin_make -DCMAKE_BUILD_TYPE=Release
...

### Expected Result

#### Running command: "make -j4 -l4" in "/home/akumar/catkin_ws/build"

### Actual Result

```
catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:4383:110: error: ‘HUMAN_BODY_MEDIUM’ is not a member of ‘sl::DETECTION_MODEL’
         if (mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_ACCURATE || mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_MEDIUM || mObjDetModel == sl::DETECTION_MODEL::HUMAN_BODY_FAST) {
                                                                                                              ^~~~~~~~~~~~~~~~~
zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:4995: recipe for target 'zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

### ZED Camera model

ZED2

### Environment

```shell
Os: Ubuntu 18.04.3 LTS
Codename:	bionic
CPU : ARMv8 Processor rev 1 (v8l) × 4 
Graphics : NVIDIA Tegra X1 (nvgpu)/integrated 
ZED SDK Version: 3.2 
ROS : ROS Melodic
```

Myzhar, 2022-05-04:
Hi @Amit10311 
with the latest ZED ROS Wrapper you must use the latest ZED SDK, which is v3.7.2: 
https://www.stereolabs.com/developers/release/

Install the new SDK, delete the `build` and `devel` folders in the root of your catkin workspace to clean the cache of CMake and compile again.

closed 2022-05-04 by Myzhar

Amit10311, 2022-05-04:
Hello, I tried with `ZED SDK v3.7.2` it has `CUDA 10.2` and my system works with `CUDA 10` it doesn't support it with ZED SDK v3.7.2.

Myzhar, 2022-05-04:
Please write an email to support@stereolabs.com and add the report file generated by the ZED_Diagnostic command.
In this way, I can get more information about your system and provide you with more pèrecise support.