stereolabs/zed-ros-wrapper #837: RGB and depth framerate drops when positional tracking starts

cedricgoubard, 2022-05-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello, 

I am trying to use the ZED2 camera to get RGB, depth and odometry data for a SLAM algorithm running on a Jetson Xavier NX. However, it looks like as soon as the positional tracking starts, both RGB and depth topics fall to ~3Hz

I tried disabling the positional tracking; I then get a good framerate for the RGB topic, but if I try to `rostopic hz` the depth topic, the positional tracking starts (even though I disable it) and all topics drop to 3Hz.  

### Steps to Reproduce

<details>
<summary>Here is the launch file</summary>

```xml

<node name="zed2_node" pkg="zed_wrapper" type="zed_wrapper_node" output="screen" required="true">
        <rosparam file="$(find MYPACKAGE)/config/zed_params/common.yaml" command="load" />
        <rosparam file="$(find MYPACKAGE)/config/zed_params/zed2.yaml" command="load" />

        <param name="general/camera_name"            value="base_zed2" />
        <param name="general/base_frame"                value="base_zed2_base_link" />
        <param name="general/zed_id"                        value="0" />
        <param name="general/gpu_id"                        value="-1" />
</node>
```

</details>

<details>
<summary>Here is the common.yaml param file</summary>

```yaml
# params/common.yaml
# Common parameters to Stereolabs ZED and ZED mini cameras
---

# Dynamic parameters cannot have a namespace
brightness:                 4                                   # Dynamic
contrast:                   4                                   # Dynamic
hue:                        0                                   # Dynamic
saturation:                 4                                   # Dynamic
sharpness:                  4                                   # Dynamic
gamma:                      8                                   # Dynamic - Requires SDK >=v3.1
auto_exposure_gain:         true                                # Dynamic
gain:                       100                                 # Dynamic - works only if `auto_exposure_gain` is false
exposure:                   100                                 # Dynamic - works only if `auto_exposure_gain` is false
auto_whitebalance:          true                                # Dynamic
whitebalance_temperature:   42                                  # Dynamic - works only if `auto_whitebalance` is false
depth_confidence:           30                                  # Dynamic
depth_texture_conf:         100                                 # Dynamic
pub_frame_rate:             15.0                                # Dynamic - frequency of publishing of video and depth data
point_cloud_freq:           10.0                                # Dynamic - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)

general:
    camera_name:                zed                             # A name for the camera (can be different from camera model and node name and can be overwritten by the launch file)
    zed_id:                     0
    serial_number:              0
    resolution:                 2                               # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
    grab_frame_rate:            15                              # Frequency of frame grabbing for internal SDK operations
    gpu_id:                     -1
    base_frame:                 'zed2_base_link'                     # must be equal to the frame_id used in the URDF file
    verbose:                    false                           # Enable info message by the ZED SDK
    svo_compression:            2                               # `0`: LOSSLESS, `1`: AVCHD, `2`: HEVC
    self_calib:                 true                            # enable/disable self calibration at starting
    camera_flip:                false
[... 283 lines cut ...]
[ INFO] [1651742997.645896751]: Expected cycle time: 0.005000000 - Real cycle time: 0.302025790
[ INFO] [1651742998.856181369]: Sensors thread is not synchronized with the Sensors rate
[ INFO] [1651742998.857478054]: Expected cycle time: 0.005000000 - Real cycle time: 0.302615170
[ INFO] [1651743000.065413489]: Sensors thread is not synchronized with the Sensors rate
[ INFO] [1651743000.066732158]: Expected cycle time: 0.005000000 - Real cycle time: 0.302426974
```

</details>

### ZED Camera model

ZED2

### Environment

```shell
Dockerised Ubuntu 20 on Jetson Xavier NX (ROS Noetic)
ZED SDK 3.7
```

### Anything else?

Apart from that, the ZED2 is a great camera :smile: 

Thank you for your help!

Myzhar, 2022-05-05:
Hi @cedricgoubard 
there is an interesting warning:
`[ WARN] [1651742971.624700659]:  !!! Failed to get static transforms. Is the 'ROBOT STATE PUBLISHER' node correctly working? `
indeed I cannot see the logs of the Robot State Publisher node that broadcasts the static TFs of the camera.
The 3 Hz problem is because the Positional Tracking module always waits for valid TF transformations before starting the elaborations.

Did you disable the URDF?

cedricgoubard, 2022-05-05:
Hi @Myzhar,
Thank you for replying so quickly!
I put the state publisher back in, and it fixed the issue :+1:

closed 2022-05-05 by cedricgoubard