stereolabs/zed-ros-wrapper #842: Questions about the resolution

canglangzhige, 2022-06-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi. When I run an API like ./ZED_Explorer, I can choose the resolution between "HD2K" "HD1080" "HD720" "VGA"：
[image]
 But when I use ROS to "run roslaunch zed_wrapper zed2.launch "(for ZED2) "roslaunch zed_wrapper zed2i.launch" (for ZED2i), use "rostopic echo /zed2i/zed_node/rgb_raw/camera_info"  or " rostopic echo /zed2/zed_node/rgb_raw/camera_info " to print camera parameters , the resolution is not normal. I choose resolution between "# '0': HD2K, '1': HD1080, '2': HD720, '3': VGA" in /zed-ros-wrapper/zed_wrapper/params/common.yaml, when When I select "2", the output camera resolution of " rostopic echo /zedi/zed_node/rgb_raw/camera_info" is (640 360):
[image]
But the resolution of the normal HD720 should be (1280 720), which means that the terminal prints out half of the normal resolution. The same is true for other resolutions, which is a problem with both the ZED2 and ZED2i cameras.

### Steps to Reproduce

1. Modify the resolution in common.yaml to HD720 (parameter is 2)
2. roslaunch zed_wrapper zed2.launch  or roslaunch zed_wrapper zed2i.launch
3. rostopic echo /zed2/zed_node/rgb_raw/camera_info or rostopic echo /zed2i/zed_node/rgb_raw/camera_info
...

### Expected Result

When I select the resolution HD720 in common.yaml, run "rostopic echo /zed2/zed_node/rgb_raw/camera_info",  the output resolution is (1280 720), also when I select the resolution as VGA, the output resolution is (672 376).

### Actual Result

When I run an API like ./ZED_Explorer, I can choose the resolution between "HD2K" "HD1080" "HD720" "VGA"：
[image]
 But when I use ROS to "run roslaunch zed_wrapper zed2.launch "(for ZED2) "roslaunch zed_wrapper zed2i.launch" (for ZED2i), use "rostopic echo /zed2i/zed_node/rgb_raw/camera_info"  or " rostopic echo /zed2/zed_node/rgb_raw/camera_info " to print camera parameters , the resolution is not normal. I choose resolution between "# '0': HD2K, '1': HD1080, '2': HD720, '3': VGA" in /zed-ros-wrapper/zed_wrapper/params/common.yaml, when When I select "2", the output camera resolution of " rostopic echo /zedi/zed_node/rgb_raw/camera_info" is (640 360):
[image]
But the resolution of the normal HD720 should be (1280 720), which means that the terminal prints out half of the normal resolution. The same is true for other resolutions, which is a problem with both the ZED2 and ZED2i cameras.

### ZED Camera model

ZED2

### Environment

```shell
OS:Ubuntu 20.04
CPU:Intel® Core™ i7-8750H CPU @ 2.20GHz × 12
GPU: NVIDIA GEFORCE RTX 2060
CUDA-11.5 
ZED SDK version: v3.7.2
other: ROS neotic
```

Myzhar, 2022-06-06:
Hi @canglangzhige 
this is not a bug, you must simply change the value of the [`video/img_downsample_factor` parameter](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L36)

canglangzhige, 2022-06-06:
> 

Thank you! I have solved it.

closed 2022-06-06 by Myzhar

canglangzhige, 2022-06-07:
> Hi @canglangzhige this is not a bug, you must simply change the value of the [`video/img_downsample_factor` parameter](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L36)

Hello, I run ./ZED_Explorer, and I can find some calibration data . As shown in the following figure:
[image]
But there is no coordinate system transformation of the camera relative to the imu. How can I query the factory parameters of the camera's coordinate system transformation relative to the imu?
In addition, what are the steps for automatic calibration of ZED2?
The parameters I calibrated by the Kalibr method are very different from the factory parameters. How much will the automatic calibration differ from the factory parameters?
If I don't calibrate and use the factory parameters directly, will it have a big impact on my results?

Myzhar, 2022-06-07:
Hi @canglangzhige 
please do not reply to a closed issue.
This is not a ZED ROS Wrapper bug, but an "information request", so please post the question to the [Stereolabs community forum](https://community.stereolabs.com/) or send an email to support@stereolabs.com