stereolabs/zed-ros-wrapper #843: zed2i framerate drops when long time recording [using rosbag record]

arenas7307979, 2022-06-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

device : zed2i

env : disable all depth and slam topic, only record stereo image and imu data

problem : missing topic data (drop frame or imu data) when rosbag record image 60hz and imu 400hz  1hour

[image]

zoom_in for check fps:
biggest difference ：60hz --> 10hz / 5hz
[image]
[image]

### Steps to Reproduce

1.roslaunch zed_wrapper zed2i.launch 
2.rosbag record -b 4096 /zed2i/zed_node/left_raw/image_raw_gray /zed2i/zed_node/right_raw/image_raw_gray /zed2i/zed_node/imu/mag /zed2i/zed_node/imu/data_raw /zed2i/zed_node/imu/data

### Expected Result

images 40hz ~ 60hz
imu ~ 400hz 

### Actual Result

zoom_in for check fps:
biggest difference ：60hz --> 10hz / 5hz
[image]
[image]

### ZED Camera model

ZED2i

### Environment

```shell
cpu:intel i7 11gen
ram:32g
```

Myzhar, 2022-06-06:
Hi @arenas7307979 
this is not a ZED ROS Wrapper bug, but it's a problem with rosbag recording.
While rosbag records data the internal rosbag buffer increases if the disk is not fast enough to save all the information while it's made available and the system slows down.
Have you tried to perform the recording on a remote machine?

arenas7307979, 2022-06-10:
Hi, @Myzhar 
I only tried to record data on the local PC
Is there any way to reduce this problem？

closed 2022-08-13 by arenas7307979