stereolabs/zed-ros-wrapper #848: [ WARN] [1658229133.787347862]: area_memory_db_path [/home/esnode/.ros/zed_area_memory.area] doesn't exist or is unreachable.

ES-3508, 2022-07-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hey, I'm using a ZED 2 with the zed_ros_wrapper on an Nvidia Jetson Nano.

When I follow RGB-D Handheld Mapping tutorial to below command

I have get this **[ WARN] [1658229133.787347862]: area_memory_db_path [/home/esnode/.ros/zed_area_memory.area] doesn't exist or is unreachable.** 

### Steps to Reproduce

1.roslaunch zed_wrapper zed2i.launch

### Expected Result

No warning

### Actual Result

[ WARN] [1658229133.787347862]: area_memory_db_path [/home/esnode/.ros/zed_area_memory.area] doesn't exist or is unreachable.

### ZED Camera model

ZED2i

### Environment

```shell
Os: Ubuntu 18.04 LTS
Jetson nano
```

Myzhar, 2022-07-19:
That's not a bug. It's a warning because the [Area Memory is enabled](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L54) but you have not yet saved a local map by calling the `saveAreaMap` service of by setting [save_area_memory_db_on_exit](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L53) to true.

closed 2022-07-19 by Myzhar

ES-3508, 2022-07-20:
So,  i have to edit  common.yaml as **save_area_memory_db_on_exit: true** ? 
I changed but still same warning are there!

ES-3508, 2022-07-20:
@Myzhar Thank you