stereolabs/zed-ros-wrapper #851: Incompatible with leggedrobotics/darknet_ros on Ubuntu20.04 and Noetic

RustSon, 2022-07-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

when I run zed wrapper and darknet_ros at the same time, a few minutes later, zed wrapper node would crash. And if I don't plug and unplug the zed camera again，zed wrapper couldn't work.

### Steps to Reproduce

1. roslaunch zed_wrapper zed.launch
2. roslaunch darknet_ros darknet_ros.launch

### Expected Result

zed wrapper and darknet_ros can work stably

### Actual Result

a few minutes later, zed wrapper crash

### ZED Camera model

ZED

### Environment

```shell
Ubuntu20.04
i7-11700K
memory:64G
RTX3070
ZED SDK: 3.7.4
ROS: Noetic
CUDA:11.3
```

### Anything else?

darknet_ros : https://github.com/leggedrobotics/darknet_ros/tree/master

Myzhar, 2022-07-26:
Hi @RustSon 
what is the error generated when the ROS node crashes?

Myzhar, 2022-07-26:
I could run the nodes correctly.

In one terminal (I'm using a ZED2i now):
`$ roslaunch zed_wrapper zed2i.launch`

In a second terminal:
`$ roslaunch darknet_ros darknet_ros.launch image:=/zed2i/zed_node/rgb/image_rect_color`

Result:
[image]

There are surely many parameters to tune (e.g. red and blue channels must be flipped), but it worked with no issues for long time.

RustSon, 2022-07-28:
Sorry! I found that there was a problem with the USB3.0 port of my computer, which led to this bug. Thank you again for your help!

closed 2022-07-28 by Myzhar