stereolabs/zed-ros-wrapper #852: Width and height is different for the pointcloud and image stream

KristianSem, 2022-07-29:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The width and height is different for the pointcloud and image stream for my zed2 camera. Is this a result of camera calibration? In that case why is the resolution different?

Pointcloud resolution: 576x1024

Camera resolution (rgb/image_rect): 503x896

My calibration params:
[LEFT_CAM_HD]
cx=634.7693
cy=357.5304
fx=715.2219
fy=714.7376
k1=0.1737
k2=0.7757
k3=-0.6430
p1=-0.0007915
p2=0.0004861

[RIGHT_CAM_HD]
cx=648.4723
cy=364.1165
fx=717.6800
fy=717.6045
k1=0.1547
k2=0.8843
k3=-0.7793
p1=0.0014
p2=0.0003

### Steps to Reproduce

1. Put your camera in a plexiglass case
2. Calibrate it with a checkerboard
3. 
...

### Expected Result

Same resolution on the pointcloud and camera stream

### Actual Result

Different resolution on the pointcloud and camera stream

### ZED Camera model

ZED2

### Environment

```shell
Jetpack 4
Jetson Xavier
```

Myzhar, 2022-07-30:
Hi @KristianSem 
what are the values of your parameters [`video/img_downsample_factor`](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L36) and [`depth/depth_downsample_factor`](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L44)?

If they are different then the final dimensions of the RGB data and depth data are different.

KristianSem, 2022-07-30:
My maann. That was the issue. Someone managed to set them to different values when fiddling around. Thanks for the quick reply

closed 2022-07-30 by KristianSem