stereolabs/zed-ros-wrapper #854: Barometer units

arenas7307979, 2022-08-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

hii, 
about  zed-ros-wrapper
what is units for topic of /zed2i/zed_node/atm_press 
[image]
[image]

 testing on the second floor below ground, and the result  doesn't look right

yhabib29, 2022-08-15:
It seems to be converted in Pascal here [zed_wrapper_nodelet.cpp:2964](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L2964)

arenas7307979, 2022-08-17:
The chart is my downstairs data. It looks like the unit should be Pascal*100, but the press change of ZED2i is very bad. The chart on the right side is the barometer of my phone

[image]

Myzhar, 2022-10-13:
Hi @arenas7307979 
the ROS units for the pressure is `Pa`, according to the[ topic definition](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/FluidPressure.html).
I agree that there is a bug in units conversion, we're fixing it.

Regarding the measurement. The sensor is inside a closed case, so you must take it into consideration.

closed 2022-10-13 by Myzhar