stereolabs/zed-ros-wrapper #858: ZED2 depth modes with AGX Xavier

tugbakara, 2022-08-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi, when I run ZED nodes, or depth viewer from SDK when we change depth mode our AGX is shutting down. Running launch with depth mode with performance is okay but out pointcloud datas in RVIZ floating they are not straight.

### Steps to Reproduce

1. dd
2.
3.
...

### Expected Result

We are ecpecting depth mode can be changable and executable in AGX.

### Actual Result

ddd

### ZED Camera model

ZED2

### Environment

```shell
OS : Ubuntu 18.04 LTS
CPU : Arm64
GPU: Nvidia Jetson Xavier AGX
ZED SDK Version: 3.7.6
ROS Melodic
```

Myzhar, 2022-08-23:
Hi @tugbakara 
the Xavier AGX and the ZED2 support all the available depth modes.
I suppose you have a power or a thermal issue.
How's the Jetson powered?

tugbakara, 2022-08-23:
> Hi @tugbakara the Xavier AGX and the ZED2 support all the available depth modes. I suppose you have a power or a thermal issue. How's the Jetson powered?

We powered it  from LiPo (45c,4S) battery and power jack, nothing affects, AGX still closes itself.

Myzhar, 2022-08-23:
I suppose you are using a Xavier AGX devkit.
Is the battery directly connected to the board or are you using a DC-DC converter to guarantee that the power is stable?

tugbakara, 2022-09-01:
Hi, @Myzhar , unfortunately we passed other brand cam, using two different brand makes everything messed up. Thanks for support, and it was power issue, if anyone gets same trouble can make sure looking that issue :)

closed 2022-09-01 by tugbakara