stereolabs/zed-ros-wrapper #866: > Hi @pavloblindnology,

leiwanporsche, 2022-09-22:
> Hi @pavloblindnology,
> 
> The first one
> 
> > 1. Rotational matrix = Rodrigues([Rx, Ry, Rz]).
> 
> So, for instance something like this using [OpenCV Rodrigues](http://docs.opencv.org/3.2.0/d9/d0c/group__calib3d.html#ga61585db663d9da06b68e70cfbf6a1eac) : `cv::Rodrigues([Rx, Ry, Rz] /*in*/, RotMatrix3x3 /*out*/)`

Hello, I have a question about this stereo parameter. Is this the transform matrix between left camera frame and right camera frame? Or is it between optical frame?

_Originally posted by @leiwanporsche in https://github.com/stereolabs/zed-ros-wrapper/issues/39#issuecomment-1255057038_

leiwanporsche, 2022-09-22:
Hi, I have a question about the stereo parameter. Is the Stereo parameter describing the transform between left camera optical frame and the right camera optical frame? And is it the transform from right to left? 
[image]

Myzhar, 2022-09-22:
I'm sorry, but this issue is not valid because it's not a **bug** or a **feature request**, but a _support request_.
Please write an email to support@stereolabs.com or ask on the èStereolabs community forum](https://community.stereolabs.com)

closed 2022-09-22 by Myzhar

JzHuai0108, 2024-06-18:
Hi friend,

I have a very late answer. Hope you already solved this problem two years ago.
The above parameters express the transform between the left camera and the right camera as below.
```python
rotvec = np.array([RX_HD, CV_HD, RZ_HD])
R_p_L = np.array([-Baseline, TY, TZ])
R_R_L, _ = cv2.Rodrigues(rotvec)
```
Why? 
Refer to [here](https://community.stereolabs.com/t/baseline-value-in-conf-calibration-file/283) and [here](http://robots.stanford.edu/cs223b04/JeanYvesCalib/htmls/example5.html).