stereolabs/zed-ros-wrapper #868: ZED model optimization error with ZED SDK 3.8

VenkataPrasadBoson, 2022-11-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

hi Zed community

Recently we got this issue while running the ROS ZED wrapper by enabling Object Detection. As we used that feature for one of the previously developed applications it became a priority for us. while testing it after some time we faced this issue. It worked for us with our previous version 3.7.6 which is not currently available.

we are using Nvidia Xavier AGX 64GB with jetpack version 4.6.1 ubuntu 20.04.

 Please help us resolve this issue.
 **Error we got** 
```
Optimizing all AI models
Optimizing: MULTI CLASS DETECTION...
Optimized model not found, trying to download it...
 Optimizing objects_performance_2.2 \  0.3%[>                             ] 13min 17s est. left         Stack trace (most recent call last):
#10   Object "ZED_Diagnostic", at 0x434687, in 
#9    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0x7fac110e0f, in __libc_start_main
#8    Object "ZED_Diagnostic", at 0x41e3a7, in 
#7    Object "ZED_Diagnostic", at 0x435c87, in 
#6    Object "ZED_Diagnostic", at 0x46942f, in 
#5    Object "ZED_Diagnostic", at 0x468de7, in 
#4    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7fa11c4b97, in 
#3    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7fa11c73e3, in 
#2    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7fa11f6517, in 
#1    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7fa11f2d13, in 
#0    Object "/usr/local/zed/lib/libsl_ai.so", at 0x7fa11eee7c, in 
Segmentation fault (Address not mapped to object [0x8])
Segmentation fault (core dumped)
```

Regards

Venkata prasad

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

Zed-based object detection to work 

### Actual Result

Zed-based object detection has to work with different applications

### ZED Camera model

ZED2i

### Environment

```shell
OS: UBUNTU 20.04
CPU: ARM
GPU: Nvidia Xavier AGX 64gb
ZED SDK version: 3.8.1
Other Info: Ros Noetic
```

Myzhar, 2022-11-02:
Hi @VenkataPrasadBoson 
This is an issue related to the ZED SDK.
Please write an email to support@stereolabs.com adding the report file generated by the ZED Diagnostic tool.

closed 2022-11-02 by Myzhar

VenkataPrasadBoson, 2022-11-02:
I asked the same question in the zed community

Myzhar, 2022-11-02:
In that case please do never cross post.
We monitor all the support channel with the same frequency, posting the same problem multiple times does not speed-up the process.

VenkataPrasadBoson, 2022-11-02:
Noted

Thanks for the Reply @Myzhar