stereolabs/zed-ros-wrapper #870: Incorrect obstacles detected on the Depth and Pointcloud topics with some lights in FOV

ladrians, 2022-11-07:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Using a Zed2 Camera at night, depending on the light conditions, spurious obstacles are shown.

### Steps to Reproduce

Use the following cropped svo file
* [svo_20221104cut.zip](https://github.com/stereolabs/zed-ros-wrapper/files/9956046/svo_20221104cut.zip)

Will notice that the depth and Pointcloud topics detail obstacles really close to the camera but the RGB topics show there is nothing there.

We updated from SDK 3.7.1 to 3.8.1 and switched from 1:PERFORMANCE to 4:NEURAL. With the latest we notice more stability (at the cost of GPU processing) but still spurious obstacles are detected.

### Expected Result

No obstacles in from of the vehicle.

### Actual Result

A lot of noice on the PointCloud/Depth topics.

### ZED Camera model

ZED2

### Environment

```shell
Ubuntu 18 on Nvidia Jetson Xavier NX
Jetpack 4.4.1 [L4T 32.4.4]
CUDA: 10.2.89 / cuDNN: 8.0.0.180
ZED SDK version: v3.8.1 on ROS Melodic
```

### Anything else?

Using the default yaml configuration at 15hz grab_rate, PointCloud at 5hz. 
Any parameter tweak we could evaluate?

Myzhar, 2022-11-09:
Hi @ladrians 
I analyzed your SVO and I noticed the same effects.
The cause of the visual artifacts is the non-homogeneity of the artificial light source that you are using which generates big differences between the left and the right images making the feature-matching step of the stereo vision algorithm unstable.
I suggest you use a light diffuser to better distribute the light in front of the camera.

ladrians, 2022-11-09:
Hi thanks for the feedback! Our concern is that the light sources are far away and in some cases it marks point very close to the Rover, for example up to 2mts, trouble making the the navigation.

I am not following how to use a light diffuser in from of the device? If we were to use a Zed2i do you think of any improvements?

Any other recommendation?

regards,

Myzhar, 2022-11-09:
The ZED2i with polarizer can reduce the effect of the vertical component of the light, this is good for sunlight, but it can be not enough for artificial lights, according to their intrinsic features.

closed 2022-12-15 by github-actions[bot]