stereolabs/zed-ros-wrapper #871: ZED not working after boot but replugging it helps

maxpolzin, 2022-11-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The issue of https://github.com/stereolabs/zed-ros-wrapper/issues/249 occurs also on the latest software versions.

### Steps to Reproduce

1. Plugin ZED camera
2. System detects the camera
3. Restart the system
4. Only the HID interface device is detected
5. Re-plug the camera (not turning USB cable) and the camera is detected again
...

### Expected Result

The camera is detected across reboots.

### Actual Result

The camera is not detected across reboots

### ZED Camera model

ZED Mini

### Environment

```shell
Xavier NX (Seedstudio A206 (same as DevKit) or Auvidea JNX30) 
Jetpack 5.0.2 
SDK 3.8
```

maxpolzin, 2022-11-09:
I couldn't reopen https://github.com/stereolabs/zed-ros-wrapper/issues/249, thus I created this new issue.

Myzhar, 2022-11-09:
Hi @maxpolzin 
thank you for the correct behavior. Opening a new issue and tagging the closed one is indeed the right way to act.

I'm sorry, but the behavior that you describe is strictly related to the carrier board and to how it initializes the USB controller.
We provided a workaround for this on the ZED2 and ZED2i with the `reboot` command that forces a re-initialization of the camera when the USB controller is ready, but unfortunately, this cannot be used with the ZED Mini.

The only available workaround is to reset the power of the USB controller of the carrier board to force a re-initialization of all the connected devices, but this does not work with all the controller models.

You can try to adapt this tool to your requirements:
https://github.com/obraun-sl/usb_power_cycle

maxpolzin, 2022-11-09:
Thx for your prompt response @Myzhar and the clarification on the subject. I'll look into the power-cycling and note updates in this issue.

maxpolzin, 2022-11-10:
My solution is switching to our Zed2i which seems to work fine.

closed 2022-11-10 by maxpolzin