stereolabs/zed-ros-wrapper #874: Nvidia related error.

Mikor-mkr, 2022-12-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello everyone,

I am trying to use the ZED m model with ROS on Jetson Nano but I get an error regarding the ZED connection.
, I've run the SDK installation file (ZED_SDK_Tegra_L4T32.7_v3.8.2.zstd.run) until completion and then I built the catkin workspace, then I sourced the workspace and launched the zed.m file. 

I've found a [similar issue](https://github.com/stereolabs/zed-ros-wrapper/issues/559) but deleting the build and devel folders did not help me.

Any ideas?

### Steps to Reproduce

1. I've run the SDK installation file (ZED_SDK_Tegra_L4T32.7_v3.8.2.zstd.run) until completion
2. Cloned the repository on my workspace
3. catkin_make
4. roslaunch zed_wrapper zedm.launch 

### Expected Result

I expected the node to start up smoothly and publish the desired topics.

### Actual Result

The node started but the output I get is the following: 

```
[ INFO] [1669890207.084863231]:  *** Opening ZED-M...
[ INFO] [1669890207.268928840]: ZED connection -> NVIDIA DRIVER OUT OF DATE
CUDA error at Camera.cpp:121 code=35(cudaErrorInsufficientDriver) "void sl::Camera::close()" 
CUDA error at Camera.cpp:149 code=35(cudaErrorInsufficientDriver) "void sl::Camera::close()" 
CUDA error at Camera.cpp:174 code=35(cudaErrorInsufficientDriver) "void sl::Camera::close()" 
CUDA error at Camera.cpp:214 code=35(cudaErrorInsufficientDriver) "void sl::Camera::close()" 
CUDA error at Camera.cpp:219 code=35(cudaErrorInsufficientDriver) "void sl::Camera::close()" 
CUDA error at CameraUtils.hpp:773 code=35(cudaErrorInsufficientDriver) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:789 code=35(cudaErrorInsufficientDriver) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:795 code=35(cudaErrorInsufficientDriver) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:798 code=35(cudaErrorInsufficientDriver) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:800 code=35(cudaErrorInsufficientDriver) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at CameraUtils.hpp:809 code=35(cudaErrorInsufficientDriver) "void sl::ObjectsDetectorHandler::clear()" 
CUDA error at Camera.cpp:234 code=35(cudaErrorInsufficientDriver) "void sl::Camera::close()" 
[ INFO] [1669890209.269835503]: ZED connection -> NVIDIA DRIVER OUT OF DATE
```

### ZED Camera model

ZED Mini

### Environment

```shell
OS: 4.9.140-tegra
CPU: ARMv8 Processor rev 1 (v8l) × 4
GPU: NVIDIA Tegra X1 (nvgpu)/integrated (Nvidia Jetson Nano)
ZED SDK version: 3.8.2
CUDA version: CUDA 10.2
```

Myzhar, 2022-12-01:
Hi @Mikor-mkr 
this is not a bug of the ZED ROS Wrapper.
For support, you can open a thread on the [Stereolabs forum](https://community.stereolabs.com) or write an email to support@stereolabs.com

closed 2022-12-01 by Myzhar

Mikor-mkr, 2022-12-01:
Thanks for the reply.