stereolabs/zed-ros-wrapper #875: Zed2i depth framerate really slow

Alessio-Parmeggiani, 2022-12-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi, I have a problem regarding the depth computed using the Zed2i on a jetson Nano on Ubuntu 20.04 with ROS noetic.
When visualizing the 
By launching the file zed2i.launch I get about 2fps for the depth images using quality=performance when looking at the images in another computer.

Some things I noticed:
- the problem starts in the exact moment I subscribe to the depth topic using Rviz or rostopic, I guess this is when the Zed starts computing the depth.
- By using zed depth viewer, so without using ROS, I get about 30fps with depth quality set to ultra.
- When visualizing using rostopic hz the rate of the depth on the Jetson Nano I get about 13 fps
- When visualizing using rostopic hz the rate of the same depth topic on another computer I get 2 fps
- When visualizing multiple normal image topics  on another computer the visualization is smooth: about 15-20 fps
 

I already tried subscribing to the compressed topics but nothing changed.

[image]

### Steps to Reproduce

1. launch zed2i.launch on the jetson nano
2. on another pc connected to the same ROS network: rostopic hz /zed2i/zed_node/depth/depth_registered/compressedDepth
3. alternative: visualize on Rviz the topic /zed2i/zed_node/depth/depth_registered

### Expected Result

The depth frequency should be higher than 2fps, it should be the same as the one of the Jetson Nano

### Actual Result

On the jetson nano I get a publishing rate of 15 hz on the depth topic, on another computer I get a rate of 2hz

### ZED Camera model

ZED2i

### Environment

```shell
Jetson Nano
Ubuntu 20.04
ROS noetic
CUDA 10.2
Jetpack 4.6-b197
```

Myzhar, 2022-12-01:
The compressed depth is a high demanding topic.
The compression of the data takes too much elaboration power and the Jetson Nano cannot handle it.

Myzhar, 2022-12-01:
Can I ask you how can you run Ubuntu 20.04 on the Jetson Nano? The JP4 is based on Ubuntu 18.04.

Alessio-Parmeggiani, 2022-12-01:
> The compressed depth is a high demanding topic. The compression of the data takes too much elaboration power and the Jetson Nano cannot handle it.

Yeah but in the shell on the Nano I can see that the depth is being computed at 15 hz so the Nano can handle it. Instead on another machine I receive it at 2 hz and it should not be a communication issue because normal images are received at the expected rate.

> Can I ask you how can you run Ubuntu 20.04 on the Jetson Nano? The JP4 is based on Ubuntu 18.04.

I needed ROS noetic so used this files: https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image

Myzhar, 2022-12-05:
Hi @Alessio-Parmeggiani 
I'm sorry, but this is not an issue related to the ZED ROS Wrapper.
It's something related to ROS itself and to a Jetson Nano not using the default standard configuration, so we cannot analyze and test this issue.
I'm closing it, please feel free to ask to re-open it if you notice that the wrong behavior is related to a bug or a misconfiguration of the Wrapper.

closed 2022-12-05 by Myzhar

bmegli, 2022-12-22:
@Alessio-Parmeggiani 

image_transport compression happens in the node publishing image and is blocking

You may decouple it, for example using image_transport republish

```bash
# run ZED node(let)
roslaunch zed_wrapper zed2i.launch
```

```bash
# republish depth from raw (locally)
rosrun image_transport republish raw in:=/zed2i/zed_node/depth/depth_registered out:=/out/depth
```

```bash
# encoded depth frequency (locally or remotely)
rostopic hz /out/depth/compressedDepth
```

---------------------------

In above:
- you add some local communication overhead (republish)
- but compression doesn't happen in ZED node

If you have free CPU cores you should be able to get more hz

Alessio-Parmeggiani, 2022-12-24:
I partially solved the problem, i used rosrun rqt_reconfigure rqt_reconfigure to change the parameters of the topic /zed2i/zed_node/depth/depth_registered/compressedDepth.

By changing compression type from png to rvl the depth image is now received at 10 Hz (the same rate at which it is being computed), the same result can be obtained with png compression by setting png quality and quantization to a lower number (e.g. png quality=1 and quantization to 2, the quality does not decrease a lot).

These parameters can then be set in the same launch file that launches the zed camera so that everything is done automatically.