stereolabs/zed-ros-wrapper #879: Point_cloud/cloud_registered corrodinate system

amoghv17, 2023-01-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am using a ZED2i camera and was trying to implement a local planner using the /odom and /point_cloud/cloud_registered topics that is being published by the zed camera. The co-ordinate system of the point cloud is "COORDINATE_SYSTEM_RIGHT_HANDED_Y_UP"  this causes the planner to generate /cmd_vel values on a different co-ordinate than the one the robot is using. The robot uses "COORDINATE_SYSTEM_RIGHT_HANDED_Z_UP_X_FWD". 
I want to change the coordinate system of the point cloud generated by the zed camera to the second coordinate system. I was not able to find the initialization of the coordinate systems anywhere in the .yaml, .urdf and src files. Could you please tell me how to change the coordinate system of the generated point cloud only and not disrupt the other rostopic's coordinate systems? 

### Steps to Reproduce

1. Run the zed-pcl application. and you can see the Coordinate system of the zed point cloud like this - 
[image]

2.
3.
...

### Expected Result

To have the coordinate system of the pointcloud to be same as the camera tf.

### Actual Result

The coordinate system of the point cloud generated is different from the ROS coordinate confiuration
[image]

### ZED Camera model

ZED2i

### Environment

```shell
OS: UBUNTU 20.04
ZED_SDK: 3.8.2 FOR UBUNTU 20
```

Myzhar, 2023-01-21:
Hi @amoghv17 
The ZED ROS Wrapper publishes the point cloud in ROS coordinate, so X FWD, Z UP:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L233

Why do you say its `COORDINATE_SYSTEM_RIGHT_HANDED_Y_UP`?

amoghv17, 2023-01-21:
I don't think it is COORDINATE_SYSTEM_RIGHT_HANDED_Y_UP, is the point cloud generated from the zed_left_optical frame and zed_right_optical frame? because the tfs of both these are different comapred to the base_link and all the other topics of the camera. 
[image]

Myzhar, 2023-01-21:
Are you sure that your URDF is correct?

Myzhar, 2023-01-21:
> I don't think it is COORDINATE_SYSTEM_RIGHT_HANDED_Y_UP, is the point cloud generated from the zed_left_optical frame and zed_right_optical frame? because the tfs of both these are different comapred to the base_link and all the other topics of the camera. 
> [image]
> 

They reflect the ROS specification, but if you look at the frame of the point cloud message you can notice that is the correct.
You can start the Rviz demo to check it:
`ros launch zed_display_rviz display_zed2i.launch`

amoghv17, 2023-01-22:
[image]
The tf topic does not display the frame of the pointcloud. Is there any other way to look at it?

Myzhar, 2023-01-23:
I meant that if you open Rviz with the default setting you can see that the point cloud is correctly displayed in front of the camera. If the TF frame is incorrect, then the point cloud has been upside down or placed in the wrong position.
If you want to know the frame id of the point cloud messages you can start `rqt` and use the `Topic Monitor` plugin to verify the frame id field in the header of the topic.

closed 2023-02-28 by github-actions[bot]

rohitmenon86, 2023-07-09:
I am using the ZED2i camera. The depth image and the rect image have frame_id = left_camera_optical_frame but only the point cloud is is in left_camera_frame. Is there a parameter I can change to have the cloud published in the optical frame?

Myzhar, 2023-07-10:
@rohitmenon86 please do not reply to closed issues. GithHub is only for bug reporting and feature requests.
For questions please write a post to the [Stereolabs community forum](https://community.stereolabs.com/)