stereolabs/zed-ros-wrapper #885: zed2i Object Detection Error: INVALID FUNCTION CALL

jamesheatonrdm, 2023-04-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

running roslaunch zed_wrapper zed2i.launch with object detection enables gives the error: Object detection error: INVALID FUNCTION CALL.
This repeats continuously as long as the node is running:
[ERROR] [1681727374.816097548]: Object detection error: INVALID FUNCTION CALL
[ INFO] [1681727374.882696006]: *** Starting Object Detection ***
[ERROR] [1681727374.882742133]: Object detection error: INVALID FUNCTION CALL
[ INFO] [1681727374.949369186]: *** Starting Object Detection ***
[ERROR] [1681727374.949422627]: Object detection error: INVALID FUNCTION CALL
[ INFO] [1681727375.016026506]: *** Starting Object Detection ***
[ERROR] [1681727375.016087341]: Object detection error: INVALID FUNCTION CALL

### Steps to Reproduce

1. Install zed-ros-examples
2. Set object_detection.od_enabled to True in zed2i.yaml
3. Launch zed wrapper node

### Expected Result

Object detection is performed successfully

### Actual Result

Object detection error: INVALID FUNCTION CALL

### ZED Camera model

ZED2i

### Environment

```shell
Ubuntu 20.04
ROS Noetic
NVIDIA RTX A4000
```

Myzhar, 2023-04-17:
What version of the ZED SDK and ROS Wrapper are you using?

jamesheatonrdm, 2023-04-17:
ZED SDK Version 3.8.2
Unsure of ROS wrapper version - how do I find it? The changelog file has 3.8.x at the top if that helps.

Myzhar, 2023-04-17:
If you are using the latest version of the master branch, then you must upgrade to ZED SDK v4.0.x.
Otherwise if you want to use ZED SDK v3.8.2 you can use the latest release:
https://github.com/stereolabs/zed-ros-wrapper/releases/tag/v3.8.x

jamesheatonrdm, 2023-04-17:
Thanks. I have upgraded to the most recent versions of both ZED SDK and ROS Wrapper. And have installed zed-ros-examples. The error goes away, and object detection seems to be working however I still cannot see the /zed2/zed_node/obj_det/objects topic available in RVIZ. 
I can echo the topic successfully in the command line.

Is this visualisation only available in ROS2? Or am I missing something to get it to work in ROS?

Myzhar, 2023-04-17:
No, you must install the RVIZ plugin that is available in the [ZED ROS Examples](https://github.com/stereolabs/zed-ros-examples) repository

closed 2023-04-17 by jamesheatonrdm