stereolabs/zed-ros-wrapper #886: camera_id not exposed through launchfiles (zedx, zedxm)

bmegli, 2023-04-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

- `zedx.launch`
- `zedxm.launch`

Should expose `camera_id` and pass it down to `zed_camera.launch.xml`

This is necessary for multi GMSL2 camera scenario.

### Use-Case

Working with multiple GMSL2 cameras:

```bash
roslaunch zed_wrapper zedx.launch camera_id:=0
```

```bash
roslaunch zed_wrapper zedxm.launch camera_id:=1
```

### Anything else?

The fix is straightforward.

I am already using and I may supply pull request (not worth it probably for such simple change).

Myzhar, 2023-04-19:
@bmegli the parameter is now exposed:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zedx.launch#L28

Let me know if it works as expected

bmegli, 2023-04-19:
Thank you @Myzhar 

Now I can run multiple GMSL2 cameras on single system

[image]

closed 2023-04-19 by bmegli