stereolabs/zed-ros-wrapper #887: ZED-X and ZED-XM support also 1080p - clarify in configuration files

bmegli, 2023-04-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

ZED-X and ZED-XM support also 1080p (1920x1080) 

Comments on acceptable values in config files are misleading

https://github.com/stereolabs/zed-ros-wrapper/blob/ab7307bca176ee621e19d2402959a276af2b7caf/zed_wrapper/params/zedx.yaml#L7

https://github.com/stereolabs/zed-ros-wrapper/blob/ab7307bca176ee621e19d2402959a276af2b7caf/zed_wrapper/params/zedxm.yaml#L7

And should be instead

```yaml
    resolution:                 2                               # '1': HD1080 -  '2': HD1200 - '4': SVGA - '6': AUTO
```

### Use-Case

User tweaking configuration file and assuming 1080p is not supported.

1080p is standard and is required in some pipelines

### Anything else?

Both ZED-X and ZED-XM work with `resolution` set to `1` (1080P)

GabrielaHruba, 2023-04-26:
This helped me solve the `ZED connection -> INVALID RESOLUTION` error I had when I was launching the `zed_multicamera_example` with two ZED X cameras. Changing the `resolution` to `1` fixed the error.

Myzhar, 2023-04-26:
When the ZED ROS Wrapper for SDK v4.0 was released the resolution 1080p was not yet available for the ZED X.
We will add it soon.

GabrielaHruba, 2023-04-27:
Oh, I completely get it. I just wanted to mention the fix to my situation in case anyone else got the same error. Thank you for all your work! :slightly_smiling_face:

Myzhar, 2023-04-27:
Solved with https://github.com/stereolabs/zed-ros-wrapper/commit/a5a6048cdf99e6159fe81a3e44cb66d07ced2686