stereolabs/zed-ros-wrapper #889: Bad magnetometer calibration with incomplete roll and pitch

flabrosse, 2023-05-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

We have zed 2i cameras mounted on 400kg robots.  These cannot be rotated al the way along the roll and pitch directions.  I have used other IMUs that were happy (or at least seemed happy) with only partial data to fit an ellipse on.  Here, the whole ellipse is fitted on only parts of the data, resulting to a bad calibration.

### Steps to Reproduce

Don't roll or pitch fully the robot.  The ellipse is fitted (badly) to the data available.

### Expected Result

The ellipse should be fitted with the data only covering a part of the ellipse, rather the whole ellipse.

### Actual Result

See expected result.

### ZED Camera model

ZED2i

### Environment

```shell
ROS noetic on ubuntu 20.04
```

### Anything else?

The other IMU was processed using rtimulib (https://github.com/RTIMULib/RTIMULib2).

I guess this is not a trivial problem, but a real one for larger robots.

Myzhar, 2023-05-03:
Hi @flabrosse 
the ZED SDK indeed performs a 3D calibration of the IMU. In your case, it can lead to unexpected behaviors.
I suggest you use the raw magnetometer data (you can use the ZED Sensor Viewer to reset the calibration) and try to use other external calibration tool.

flabrosse, 2023-05-03:
OK.  I'll have to investigate.  Is the format of the calibration file documented somewhere?

closed 2023-06-08 by github-actions[bot]