stereolabs/zed-ros-wrapper #896: Warning to reduce frame rate

ashBabu, 2023-05-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Same issue as #555. `Elaboration takes longer (0.0236713 sec) than requested by the FPS rate (0.016666667 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.`  No RViz. No changes in anything. Just cloned the repos. The output of jetson clock is as follows
```
SOC family:tegra234  Machine:Jetson AGX Orin
Online CPUs: 0-11
cpu0: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2035200 IdleStates: WFI=1 c7=1 
cpu1: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu10: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu11: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu2: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2188800 IdleStates: WFI=1 c7=1 
cpu3: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2188800 IdleStates: WFI=1 c7=1 
cpu4: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu5: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu6: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu7: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu8: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
cpu9: Online=1 Governor=schedutil MinFreq=729600 MaxFreq=2201600 CurrentFreq=2201600 IdleStates: WFI=1 c7=1 
GPU MinFreq=306000000 MaxFreq=1300500000 CurrentFreq=1300500000
EMC MinFreq=204000000 MaxFreq=3199000000 CurrentFreq=3199000000 FreqOverride=0
DLA0_CORE:   Online=1 MinFreq=0 MaxFreq=1600000000 CurrentFreq=1600000000
DLA0_FALCON: Online=1 MinFreq=0 MaxFreq=844800000 CurrentFreq=844800000
DLA1_CORE:   Online=1 MinFreq=0 MaxFreq=1600000000 CurrentFreq=1600000000
DLA1_FALCON: Online=1 MinFreq=0 MaxFreq=844800000 CurrentFreq=844800000
PVA0_VPS0: Online=1 MinFreq=0 MaxFreq=1152000000 CurrentFreq=1152000000
PVA0_AXI:  Online=1 MinFreq=0 MaxFreq=832000000 CurrentFreq=832000000
FAN Dynamic Speed control=active hwmon2_pwm=93
NV Power Mode: MAXN
```

### Steps to Reproduce

`roslaunch zed_wrapper zedxm.launch`.

### Expected Result

No Warning msg ``Elaboration takes longer (0.0236713 sec) than requested by the FPS rate (0.016666667 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.`

### Actual Result

Produces a warniing

### ZED Camera model

ZED Mini

### Environment

```shell
Nvidia jetson AGX Orin 
ZedX mini camera
```

### Anything else?

The camera model is not mentioned in the dropdown

ladrians, 2023-06-03:
Same issue here

```
Elaboration takes longer (0.0713881 sec) than requested by the FPS rate (0.066666667 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.
```

I am running on `noetic` version with a Jetson Xavier NX and just updated zed SDK to 4.0.3.

Already tweaked some frame rate for the image and pointcloud topics, but still cannot lower the `grab_frame_rate` which is making the warning appear. Part of the configuration is:

```
[ INFO] [1685819787.121696150]: SDK version : 4.0.3
[ INFO] [1685819787.121839860]: *** GENERAL PARAMETERS ***
[ INFO] [1685819787.124554904]:  * Camera Model by param        -> zed2
[ INFO] [1685819787.126946975]:  * Camera Name                  -> zed
[ INFO] [1685819787.129841345]:  * Camera Resolution            -> HD720
[ WARN] [1685819787.132781122]: Wrong FrameRate (5) for the resolution HD720. Set to 15 FPS.
[ INFO] [1685819787.132966688]:  * Camera Grab Framerate        -> 15
...
[ INFO] [1685819787.411921729]:  * [DYN] Depth confidence       -> 50
[ INFO] [1685819787.414487462]:  * [DYN] Depth texture conf.    -> 100
...
[ INFO] [1685819787.416456177]:  * [DYN] pub_frame_rate         -> 5 Hz
[ INFO] [1685819787.419451986]:  * [DYN] point_cloud_freq       -> 5 Hz
...
```
My guess is that if I could drop it, the whole nodelet consumption would be much lower. For my usecase I guess 5hz would be enough for camera topics.

Any chance to do some of these configuration changes apply?

ladrians, 2023-07-04:
Just wanted to know if there a prevision to check this case?

karmenlu, 2023-08-04:
Noticed this reduce frame rate warning yesterday as well. Also, running `roslaunch zed_wrapper zedxm.launch` with the Jetson Orin AGX and a ZED X Mini.

Myzhar, 2023-10-06:
Fixed with latest merge https://github.com/stereolabs/zed-ros-wrapper/pull/918

closed 2023-10-06 by Myzhar