stereolabs/zed-ros-wrapper #898: Build issue in ROS Melodic

Nareshkumar-g, 2023-06-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi, i am using Zed box xavier NX with zed 2 camera, i tried building zed wrapper package but the build failed and i got the below mentioned issue,
[ 61%] Generating EusLisp code from zed_interfaces/Skeleton3D.msg
Scanning dependencies of target ZEDSyncTest
Traceback (most recent call last):
  File "/opt/ros/melodic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 50, in <module>
    sys.argv, msg_template_map, srv_template_map)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 99, in _generate_msg_from_file
    search_path)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: 'module' object has no attribute 'Interpreter'
zed-ros-interfaces/CMakeFiles/zed_interfaces_generate_messages_cpp.dir/build.make:89: recipe for target '/home/productarmor/naresh_ws/devel/include/zed_interfaces/Skeleton3D.h' failed
make[2]: *** [/home/productarmor/naresh_ws/devel/include/zed_interfaces/Skeleton3D.h] Error 1
make[2]: *** Deleting file '/home/productarmor/naresh_ws/devel/include/zed_interfaces/Skeleton3D.h'
make[2]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/opt/ros/melodic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 50, in <module>
    sys.argv, msg_template_map, srv_template_map)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 99, in _generate_msg_from_file
    search_path)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: 'module' object has no attribute 'Interpreter'
zed-ros-interfaces/CMakeFiles/zed_interfaces_generate_messages_cpp.dir/build.make:105: recipe for target '/home/productarmor/naresh_ws/devel/include/zed_interfaces/PlaneStamped.h' failed
make[2]: *** [/home/productarmor/naresh_ws/devel/include/zed_interfaces/PlaneStamped.h] Error 1
make[2]: *** Deleting file '/home/productarmor/naresh_ws/devel/include/zed_interfaces/PlaneStamped.h'
Traceback (most recent call last):
  File "/opt/ros/melodic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 50, in <module>
    sys.argv, msg_template_map, srv_template_map)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 99, in _generate_msg_from_file
    search_path)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: 'module' object has no attribute 'Interpreter'
zed-ros-interfaces/CMakeFiles/zed_interfaces_generate_messages_cpp.dir/build.make:119: recipe for target '/home/productarmor/naresh_ws/devel/include/zed_interfaces/RGBDSensors.h' failed
make[2]: *** [/home/productarmor/naresh_ws/devel/include/zed_interfaces/RGBDSensors.h] Error 1
make[2]: *** Deleting file '/home/productarmor/naresh_ws/devel/include/zed_interfaces/RGBDSensors.h'
CMakeFiles/Makefile2:4040: recipe for target 'zed-ros-interfaces/CMakeFiles/zed_interfaces_generate_messages_cpp.dir/all' failed
make[1]: *** [zed-ros-interfaces/CMakeFiles/zed_interfaces_generate_messages_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 62%] Building CXX object zed-ros-wrapper/zed_wrapper/CMakeFiles/zed_wrapper_node.dir/src/zed_wrapper_node.cpp.o
[ 63%] Generating EusLisp code from zed_interfaces/PlaneStamped.msg
[ 63%] Generating EusLisp code from zed_interfaces/RGBDSensors.msg
[ 64%] Building CXX object zed-ros-examples/tests/zed_sync_test/CMakeFiles/ZEDSyncTest.dir/src/rgbd_test_sync.cpp.o
[ 65%] Generating EusLisp code from zed_interfaces/Keypoint3D.msg
[ 65%] Building CXX object zed-ros-examples/rviz-plugin-zed-od/CMakeFiles/rviz_plugin_zed_od.dir/src/plugin/src/zed_od_display.cpp.o
In file included from /home/productarmor/naresh_ws/src/zed-ros-examples/rviz-plugin-zed-od/src/plugin/src/zed_od_display.cpp:25:0:
/home/productarmor/naresh_ws/src/zed-ros-examples/rviz-plugin-zed-od/src/plugin/include/zed_od_display.hpp:28:10: fatal error: zed_interfaces/ObjectsStamped.h: No such file or directory
 #include <zed_interfaces/ObjectsStamped.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
zed-ros-examples/rviz-plugin-zed-od/CMakeFiles/rviz_plugin_zed_od.dir/build.make:62: recipe for target 'zed-ros-examples/rviz-plugin-zed-od/CMakeFiles/rviz_plugin_zed_od.dir/src/plugin/src/zed_od_display.cpp.o' failed
make[2]: *** [zed-ros-examples/rviz-plugin-zed-od/CMakeFiles/rviz_plugin_zed_od.dir/src/plugin/src/zed_od_display.cpp.o] Error 1
CMakeFiles/Makefile2:5198: recipe for target 'zed-ros-examples/rviz-plugin-zed-od/CMakeFiles/rviz_plugin_zed_od.dir/all' failed
make[1]: *** [zed-ros-examples/rviz-plugin-zed-od/CMakeFiles/rviz_plugin_zed_od.dir/all] Error 2

### Steps to Reproduce

1.
2.
3.
...
Complete build

### Expected Result

100% build

### Actual Result

80% build

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 18.04
Ros version : Ros Melodic
CUDA : 10.2
ZED_SDK : 4.0
```

Myzhar, 2023-06-12:
@Nareshkumar-g did you correctly install the ROS node? It seems that you did not pull the submodule as described here:
https://github.com/stereolabs/zed-ros-wrapper#build-the-repository

Nareshkumar-g, 2023-06-12:
@Myzhar  Tq for your support, I pulled the submodule as described,
I also checked now its updated.

Fetching submodule zed-ros-interfaces
Already up to date.

Myzhar, 2023-06-12:
Please check that the zed-interfaces folder is populated.
If you cloned the repository wrongly then the pull command will not work correctly

Nareshkumar-g, 2023-06-13:
@Myzhar Hi tq for your support,
I removed the entire package and reinstalled from the first and got core dumped error,
Then i downgraded the version of numpy and it works,
The issue is solved.

closed 2023-06-13 by Nareshkumar-g