stereolabs/zed-ros-wrapper #910: rostopic image error!!!

Superyanzhuang, 2023-08-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Thank you for your open source work. It's great. I encountered a problem using:
I set the resolution of zed2i.yaml to H2K
But the ROSTOPIC echo output is:
< Frame_ Id: "zed2i_right_camera_optical_frame"
Height: 621
Width: 1104
Encoding: "bgra8"
Is_ Bigendian: 0
Step: 4416
Data: "<array type: uint8, length: 2742336>
>
Why wouldn't the width and height of an image be 1280x720??
I use the zed-sdk API to read 1280x720? It's very strange, how should I set up Rotopic to output 1280x720 images??

### Use-Case

ow should I set up Rotopic to output 1280x720 images??

Patrick-AA, 2023-08-08:
Hello,
In the common.yaml file you can set the [img_downsample_factor](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L34) to 1.0 to match the resolution you want. This parameter rescales the output image before publishing it on ROS topics.

You can refer to [this page](https://www.stereolabs.com/docs/video/camera-controls/) to check the expected resolutions.

Regards,
Patrick

closed 2023-08-08 by Patrick-AA

Superyanzhuang, 2023-08-08:
> Hello, In the common.yaml file you can set the [img_downsample_factor](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L34) to 1.0 to match the resolution [...]

Thank you very much for your feedback. The problem has been resolved