stereolabs/zed-ros-wrapper #916: publish tf for odom and map seems to be bugged with nan transform

khuechuong, 2023-09-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I ran zed wrapper for zedm and I get this error:
        
 at line 240 in /tmp/binarydeb/ros-noetic-tf2-0.7.6/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "base_link" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)

I ran rostopic echo tf and I see this:

transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1694649918
        nsecs: 713624714
      frame_id: "odom"
    child_frame_id: "base_link"
    transform: 
      translation: 
        x: nan
        y: nan
        z: nan
      rotation: 
        x: nan
        y: nan
        z: nan
        w: nan

### Steps to Reproduce

1.roslaunch zed_wrapper zedm.launch
2.rostopic echo tf
3.
...

### Expected Result

normal run 

### Actual Result

nan transform error non stop

### ZED Camera model

ZED Mini

### Environment

```shell
ROS Noetic
Ubuntu 20.04
```

Myzhar, 2023-09-14:
Hi @khuechuong is your camera working correctly?
Can you open it with ZED Sensor Viewer and let me know if you can currently acquire inertial data?

khuechuong, 2023-09-14:
After a while, I just decided to uninstall and reinstall plus reboot and that solved it.

closed 2023-09-14 by khuechuong