stereolabs/zed-ros-wrapper #922: Using Zed2 with Kalibr

ManChrys, 2023-10-09:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I want to record a bag file to use it for Kalibr package as you described here [https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-](url) .

### Steps to Reproduce

1.roslaunch zed_wrapper zed2.launch
2.rosbag record -O Kalibr_data.bag /zed2/zed_node/imu/data_raw /zed2/zed_node/left/image_rect_color /zed2/zed_node/right/image_rect_color
3.rqt_bag Kalibr_data.bag

### Expected Result

I was expecting something like that

[image]

### Actual Result

but i have this result. i tried other topics too but it has the same results.

[image]

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 18.04.06 LTS
CPU:AMD® Ryzen 7 3700x 8-core processor × 16 
GPU:NVIDIA GeForce RTX 2060 SUPER
ZED SDK VERSION:4.0.7
```

### Anything else?

I have 2 zed2 cameras and they do the same. Thats how it works or something went wrong ?

Thanks in advance.

Myzhar, 2023-10-09:
@ManChrys it's expected. Probably the rosbag player does not support the 32bit format of the ZED Wrapper image topics, but only 24bit format.
The data are available in any case and you can perform the calibration.

ManChrys, 2023-10-09:
Ok i see. Thank yoy my friend !!

closed 2023-10-09 by ManChrys