stereolabs/zed-ros-wrapper #924: 'zed2i/zed_node/point_cloud/cloud_registered' topic frequency

wjd123ap, 2023-10-20:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I changed both grab rate and point cloud frequency in  common.yaml and zed2i.yaml. 'zed2i/zed_node/point_cloud/cloud_registered' topic frequency remains unchanged and is published fixed at 15Hz. The image is published at frequency I set.

### Steps to Reproduce

1.roslauch zed_wrapper zed2i.launch
2. rostopic hz  'zed2i/zed_node/point_cloud/cloud_registered'
3.
...

### Expected Result

Please solve the topic frequency problem of the point cloud

### Actual Result

subscribed to [/zed2i/zed_node/point_cloud/cloud_registered]
average rate: 15.438
	min: 0.062s max: 0.066s std dev: 0.00194s window: 4
average rate: 15.125
	min: 0.062s max: 0.067s std dev: 0.00114s window: 19
average rate: 15.074
	min: 0.062s max: 0.070s std dev: 0.00124s window: 34
average rate: 15.059
	min: 0.062s max: 0.070s std dev: 0.00115s window: 50
average rate: 15.052
	min: 0.062s max: 0.070s std dev: 0.00112s window: 55

### ZED Camera model

ZED2i

### Environment

```shell
OS: jetpack5.1.1
CPU:ARM
GPU:jetson orin nx 16G
sdk:4.0
other:ros noetic
```

Myzhar, 2023-10-23:
Hi @wjd123ap 
please post the values that you set for each parameter.

closed 2023-11-28 by github-actions[bot]

Song-Jingyu, 2024-01-05:
Hi, I have exactly the same issue. I set 30 for all the frame rate parameters and I tried different depth quality but all get 15Hz for the point cloud topic.

Alessio-Parmeggiani, 2024-01-08:
I have the same issue, changing the point_cloud_freq parameter in the common.yaml file does not change the publish rate for the pointcloud topic.