stereolabs/zed-ros-wrapper #933: ZED ros wrapper really slow on orin nano!

aravi-2001, 2024-02-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am running the zed ros wrapper on HD720 mode and ive set it to 60 FPS.

The output of rostopic hz /zed2i/zed_node/right/image_rect_color is only 14FPS.
The pixel resolution from numpy also is showing 360 x 640. Jetson orin nano is set on the max power mode 15W

I am sending the ros message down a long range ethernet. any leads on how to speed this would be appreciated. 

### Steps to Reproduce

1. as described above
2.
3.
...

### Expected Result

i expected the rostopic hz to show close to 60FPS atleast.

### Actual Result

14FPS for the topics

### ZED Camera model

ZED2i

### Environment

```shell
OS:UBuntu 20.04
Nvidia Jetson Orin Nano
Jetpack 5.1.1
```

Myzhar, 2024-02-23:
Hi @aravi-2001 
did you run the `jetson_clocks` script to get the maximum performance from your Orin Nano?
Are you using NEURAL depth or other depth modes?

aravi-2001, 2024-02-23:
Yes i did run the jetson_clocks script. 
And yes the depth mode is Neural

closed 2024-02-23 by Myzhar