stereolabs/zed-ros-wrapper #940: Cannot catkin build with SDK 4.1

micat001, 2024-04-10:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Using Ubuntu 20.04, CUDA 12.2, ROS 1 Noetic and the ZED SK 4.1, I can no longer catkin build the zed-ros-wrapper package.

I recently pulled the latest changes from the repo and updated my SDK to 4.1.

The issue is easily resolved by checkout out tag 4.0.8. Is there an incompatibility with Noetic I should be aware of?

### Steps to Reproduce

1. git pull zed-ros-wrapper
2. download and build latest ZED SDK
3. catkin clean and catkin build
...

### Expected Result

Successful build

### Actual Result

```bash
Errors     << zed_nodelets:make /logs/zed_nodelets/build.make.003.log                                                     
In file included from /zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:25:
zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:639:76: error: ‘GEN_’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  639 |   sl::POSITIONAL_TRACKING_MODE mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_;
      |                                                                            ^~~~
zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::readPosTrkParams()’:
zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:962:51: error: ‘GEN_1’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  962 |       mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_1;
      |                                                   ^~~~~
zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:966:51: error: ‘GEN_2’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  966 |       mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_2;
      |                                                   ^~~~~
zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:972:51: error: ‘GEN_2’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  972 |       mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_2;
      |                                                   ^~~~~
make[2]: *** [CMakeFiles/ZEDNodelets.dir/build.make:76: CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:351: CMakeFiles/ZEDNodelets.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
build/zed_nodelets; catkin build --get-env zed_nodelets | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
```

### ZED Camera model

ZED

### Environment

```shell
0S: Linux 20.04 
CPU: AMD64 
GPU: NVIDIA GeForce RTX 4080
CUDA: 12.2
ZED SDK: 4.1 
ROS Noetic
```

Myzhar, 2024-04-10:
@micat001 did you pull the latest `master` branch code and did you clean the workspace cache before compiling?
More info:
https://github.com/stereolabs/zed-ros-wrapper?tab=readme-ov-file#update-the-local-repository

micat001, 2024-04-10:
Yes, let me be more explicit in the steps and output:

1, Check local branches:
```bash
zed-ros-wrapper$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

zed-ros-interfaces$ git status
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
```
2. Checkout master branch:
```bash
zed-ros-wrapper$ git checkout master # if you are not on the main branch  
Already on 'master'
Your branch is up to date with 'origin/master'.
```
3. Update submodules:
```bash
zed-ros-wrapper$ git pull --recurse-submodules # update recursively all the submodules
Fetching submodule zed-ros-interfaces
Already up to date.
```
4. Clear build and devel logs
```bash
# i had been using catkin clean -y but just tried:
rm -rf devel/ build
```
5. Build
```bash
catkin build -DCMAKE_BUILD_TYPE=Release
```
6. Errors:
```bash
Errors     << zed_nodelets:make logs/zed_nodelets/build.make.001.log                                                 
In file included from /zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:25:
zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:639:76: error: ‘GEN_2’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  639 |   sl::POSITIONAL_TRACKING_MODE mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_2;
      |                                                                            ^~~~~
/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘void zed_nodelets::ZEDWrapperNodelet::readPosTrkParams()’:
/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:962:51: error: ‘GEN_1’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  962 |       mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_1;
      |                                                   ^~~~~
/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:966:51: error: ‘GEN_2’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  966 |       mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_2;
      |                                                   ^~~~~
/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:972:51: error: ‘GEN_2’ is not a member of ‘sl::POSITIONAL_TRACKING_MODE’
  972 |       mPosTrkMode = sl::POSITIONAL_TRACKING_MODE::GEN_2;
      |                                                   ^~~~~
make[2]: *** [CMakeFiles/ZEDNodelets.dir/build.make:76: CMakeFiles/ZEDNodelets.dir/src/zed_nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:351: CMakeFiles/ZEDNodelets.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

My Build System:
```bash
nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Aug_15_22:02:13_PDT_2023
Cuda compilation tools, release 12.2, V12.2.140
Build cuda_12.2.r12.2/compiler.33191640_0

 nvidia-smi
Wed Apr 10 10:52:09 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.161.07             Driver Version: 535.161.07   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4080        On  | 00000000:0A:00.0  On |                  N/A |
|  0%   51C    P3              40W / 320W |   2615MiB / 16376MiB |      6%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

rosversion -d
noetic

```

Myzhar, 2024-04-10:
Did you install the new ZED SDK v4.1?
https://www.stereolabs.com/developers/release#82af3640d775

The master branch is only compatible with the latest ZED SDK release

micat001, 2024-04-10:
Thanks @Myzhar - that did the trick. I had a system-wide issue with where the new SDK was installed. Removed the old, installed the new, and restarted my machine and it built just fine. Closing.

closed 2024-04-10 by micat001