stereolabs/zed-ros-wrapper #941: Positional tracking GEN_2 strange behaviors: covariance values and randomly stopping publishing messages

Alessio-Parmeggiani, 2024-04-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi, I was testing the ros wrapper with the new GEN_2 positional tracking with ZED SDK 4.1 and I noticed two strange behaviors:

The covariance values from the topic `/zedxm/zed_node/odom` are filled only on the diagonal and the values are all equal to 1.00000somethingE-10, the matrix also does not change at all during the movement of the camera. The same is true also for the  `/zedxm/zed_node/pose_with_covariance` topic.  Is this expected? 
In the previous positional tracking version (now GEN_1) the covariance matrix was completely filled and values appeared more correct,as they were all different and change based on the movement of the camera.

The second strange behavior happens randomly during the movement of the camera, in fact the camera stops publishing values on the `/zedxm/zed_node/odom` topic. I think it could happen when the camera get lost but if it is the case, it would be nice to have this published somewhere: as a warning on the terminal, or published on the `/zedxm/zed_node/odom/status` topic. 
Again, is this behavior expected?

### Steps to Reproduce

1. Change the pos_tracking_mode value in the file common.yaml with "GEN_2"
2. Start ZED node: `roslaunch zed_wrapper zedxm.launch`
3. Look at odom values: `rostopic echo -c /zedxm/zed_node/odom`

### Expected Result

I expect the odometry output to be more similar to the one produced by the GEN_1 algorithm, i.e. the covariance matrix is all filled and with values that change, and the odometry should not stop randomly.

### Actual Result

The covariance values are filled only on the diagonal and with static values.
The odom topic stops publishing messages without any warning or special status in the odom/status topic.

### ZED Camera model

ZED

### Environment

```shell
Platform: Jetson Orin Nano Developer kit
OS: Jetpack 5.1.1, Ubuntu 20.04
Camera: ZEDXM
ROS: ROS Noetic
ZED SDK: 4.1
```

Myzhar, 2024-04-15:
Hi @Alessio-Parmeggiani 

> The covariance values from the topic /zedxm/zed_node/odom are filled only on the diagonal and the values are all equal to 1.00000somethingE-10, the matrix also does not change at all during the [...]

This is an expected behavior. The covariance with GEN_2 is not yet available.

> The second strange behavior happens randomly during the movement of the camera, in fact the camera stops publishing values on the /zedxm/zed_node/odom topic. I think it could happen when the camera [...]

Can you please check the status with the Diagnostic? You can use `rqt` and the "Runtime Monitor" plugin.
I expect it to be `SEARCHING`

Alessio-Parmeggiani, 2024-04-15:
> That means indeed that the visual tracking lost the information: https://www.stereolabs.com/docs/api/group__PositionalTracking__group.html#gae481924a311113375bb3a99748d79e17

Ok, thank you, everything is clear now

closed 2024-04-15 by Alessio-Parmeggiani